start-jsk/rtmros_common #629: Indigo build failing on ROS farm

130s, 2015-01-03:
http://jenkins.ros.org/job/ros-indigo-openrtm-tools_binarydeb_trusty_i386/
It's been failing since 12/24/2014

http://jenkins.ros.org/job/ros-indigo-openrtm-tools_binarydeb_trusty_i386/12/console
```
-- catkin 0.6.11
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "openrtm_aist_python" with any of the following names:

    openrtm_aist_pythonConfig.cmake
    openrtm_aist_python-config.cmake

  Add the installation prefix of "openrtm_aist_python" to CMAKE_PREFIX_PATH
  or set "openrtm_aist_python_DIR" to a directory containing one of the above
  files.  If "openrtm_aist_python" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  catkin.cmake:7 (find_package)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!
```
`openrtm_aist_python` seems to be built fine though
http://www.ros.org/debbuild/indigo.html?q=openrtm_aist_python

k-okada, 2015-01-03:
please check https://github.com/start-jsk/rtmros_common/pull/625/files,

k-okada, 2015-01-03:
After a long discussion on
-
https://github.com/start-jsk/openrtm_aist_core/issues/1#issuecomment-35198032
- https://github.com/start-jsk/choreonoid/pull/1
and I worked on bloom release for an year, my conclusion for dealing with
non-ROS program is

- keep original repository and do not create ros-package just to download
and compile original repo
- create release repository and that will hold sourcecode (
https://github.com/tork-a/openrtm_aist-release), you can catkin compile
(catkin_make_isolated or catkin build) this pcakage
- development users will download original repository and add package.xml
from https://github.com/tork-a/openrtm_aist-release/tree/master/indigo,
this is common on non-ROS package (see https://github.com/ros/urdfdom)

this means these package is no-longer ROS-package so you cant
`find_package(catkin .... openrtm_aist)`, you have to use
`pkg_check_modules` as other packages.

k-okada, 2015-01-03:
as for full support on indigo, we have to solve
https://github.com/rdiankov/collada-dom/pull/12 problem

130s, 2015-02-03:
This is gone. Thank you @k-okada

closed 2015-02-03 by 130s