start-jsk/rtmros_common #648: Failed to compile openhrp3 with catkin tools

garaemon, 2015-02-05:
```
$ catkin b openhrp3 -i -v
==> Expanding alias 'b' from 'catkin b openhrp3 -i -v' to 'catkin build openhrp3 -i -v'
---------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/leus/ros/hydro
Source Space:       [exists] /home/leus/ros/hydro/src
Build Space:        [exists] /home/leus/ros/hydro/build
Devel Space:        [exists] /home/leus/ros/hydro/devel
Install Space:     [missing] /home/leus/ros/hydro/install
DESTDIR:                     None
---------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
---------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
---------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------- 
Found '277' packages in 0.5 seconds. 
Starting ==> openrtm_aist                                                                                                                             
[openrtm_aist] ==> '/home/leus/ros/hydro/build/openrtm_aist/build_env.sh /usr/bin/make cmake_check_build_system' in '/home/leus/ros/hydro/build/openrtm_aist' 
[openrtm_aist] <== '/home/leus/ros/hydro/build/openrtm_aist/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'        
[openrtm_aist] ==> '/home/leus/ros/hydro/build/openrtm_aist/build_env.sh /usr/bin/make -j40 -l40' in '/home/leus/ros/hydro/build/openrtm_aist'        
[openrtm_aist] Making all in build                                                                                                                    
[openrtm_aist] Making all in src                                                                                                                      
[openrtm_aist] Making all in lib                                                                                                                      
[openrtm_aist] Making all in coil                                                                                                                     
[openrtm_aist] Making all in include                                                                                                                  
[openrtm_aist] Making all in coil      
...
[openrtm_aist] Making install in .                                                                                                                    
[openrtm_aist] [100%] Built target install_openrtm_aist                                                                                               
[openrtm_aist] -- Up-to-date: /home/leus/ros/hydro/devel/share/openrtm_aist/package.xml                                                               
[openrtm_aist] <== '/home/leus/ros/hydro/build/openrtm_aist/build_env.sh /usr/bin/make install' finished with return code '0'                         
Finished <== openrtm_aist [ 15.9 seconds ]                                                                                                            
Starting ==> openhrp3                                                                                                                                 
[openhrp3] ==> '/home/leus/ros/hydro/build/openhrp3/build_env.sh /usr/bin/cmake /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_common/openhrp3 -DCMAKE_INSTALL_PREFIX=/home/leus/ros/hydro/devel' in '/home/leus/ros/hydro/build/openhrp3' 
[openhrp3] -- use C interface for Lapack                                                                                                              
[openhrp3] -- use c interaface for Blas                                                                                                               
[openhrp3] -- Found /usr/lib/atlas-base/atlas/libblas.so;/usr/lib/atlas-base/atlas/liblapack.so in /usr/lib/atlas-base/atlas                          
[openhrp3] -- Looking for Eigen - found                                                                                                               
[openhrp3] CMake Error at cmake_modules/FindOpenRTM.cmake:21 (message):                                                                               
[openhrp3]   rtm-config was not found in NOT_FOUND/bin.  Please set OPENRTM_DIR                                                                       
[openhrp3]   correctly.                                                                                                                               
[openhrp3] Call Stack (most recent call first):                                                                                                       
[openhrp3]   CMakeLists.txt:259 (find_package)                                                                                                        
[openhrp3]                                                                                                                                            
[openhrp3]                                                                                                                                            
[openhrp3] -- Configuring incomplete, errors occurred!                                                                                                
[openhrp3] <== '/home/leus/ros/hydro/build/openhrp3/build_env.sh /usr/bin/cmake /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_common/openhrp3 -DCMAKE_INSTALL_PREFIX=/home/leus/ros/hydro/devel' failed with return code '1' 
[build] There were errors:                                                                                                                            

    Failed to build package 'openhrp3' because the following command:

        # Command run in directory: /home/leus/ros/hydro/build/openhrp3
        /home/leus/ros/hydro/build/openhrp3/build_env.sh /usr/bin/cmake /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_common/openhrp3 -DCMAKE_INSTALL_PREFIX=/home/leus/ros/hydro/devel

    Exited with return code: 1 
[build] Runtime: 17.5 seconds 
```

k-okada, 2015-02-05:
humm, it worked for me on both indigo/hydro, please  cehck
- you removed existing openrtm package, apt-get remove ros-$ROS_DISTRO-openrtm-aist
- check catkin version by `rosversion catkin` 0.5.90 for hydro, 0.6.11 for indigo

garaemon, 2015-02-05:
Today I installed two environments and one works and another didn't... I will check it tomorrow

garaemon, 2015-02-06:
It's related to parallelization.
I'm testing on 40 threds machine.

catkin build openhrp3 => success
catkin build => failed

garaemon, 2015-02-06:
> you removed existing openrtm package, apt-get remove ros-$ROS_DISTRO-openrtm-aist
removed

> check catkin version by rosversion catkin 0.5.90 for hydro, 0.6.11 for indigo
need to update?
```
$ rosversion catkin
0.5.89
```

garaemon, 2015-02-06:
updated to 0.5.90 but this error still exists

k-okada, 2015-02-06:
does `[openhrp3] rtm-config was not found in NOT_FOUND/bin. Please set
OPENRTM_DIR ` happens everytime you compile ?or just randomly?
I tried with my 40 cpu machine(jenkins) and it fails as follows... `

```
Building CXX object
sample/example/simulationEC/CMakeFiles/SampleSimulationECComp.dir/SampleSimulationEC.o
Building CXX object
sample/controller/SampleCrawler/CMakeFiles/SampleCrawlerComp.dir/SampleCrawlerComp.o
Building CXX object
sample/controller/PA10Controller/CMakeFiles/PA10ControllerComp.dir/PA10ControllerComp.o
[ 14%] Building CXX object
sample/controller/SamplePD_HG/CMakeFiles/SamplePD_HGComp.dir/SamplePD_HGComp.o
[ 14%] omniidl: Cannot rename "OpenHRP/new__init__.py" to
"OpenHRP/__init__.py".
Building CXX object
sample/controller/PD_HGtest/CMakeFiles/PD_HGtest.dir/PD_HGtest.o
[ 15%] [ 16%] [ 17%] [ 17%] omniidl: Cannot rename
"OpenHRP__POA/new__init__.py" to "OpenHRP__POA/__init__.py".
Building CXX object
sample/controller/SampleCrawler/CMakeFiles/SampleCrawlerComp.dir/SampleCrawler.o
Building CXX object
sample/controller/SamplePD_HG/CMakeFiles/SamplePD_HGComp.dir/SamplePD_HG.o
make[2]: *** [hrplib/hrpCorba/Controller_idl.py] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 18%] [ 18%] Building CXX object
sample/controller/PA10Controller/CMakeFiles/PA10ControllerComp.dir/PA10Controller.o
make[2]: *** [hrplib/hrpCorba/DynamicsSimulator_idl.py] Error 1
[ 19%] Building CXX object
sample/controller/PD_HGtest/CMakeFiles/PD_HGtestComp.dir/PD_HGtest.o
````

k-okada, 2015-02-06:
humm, check `env | grep OPENRTM` before you caompile...