start-jsk/rtmros_common #650: compiling openrtm_aist second time takes much time with catkin build

garaemon, 2015-02-06:
`catkin build openrtm_aist` takes 20 seconds over even though it's been compiled before with ccache:

```
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ catkin build openrtm_aist
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
Found '204' packages in 1.2 seconds. 
Starting ==> openrtm_aist                                                                                   
Finished <== openrtm_aist [ 25.0 seconds ]                                                                  
[build] Finished.                                                                                           
[build] Runtime: 26.7 seconds 
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ catkin build openrtm_aist
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
Found '204' packages in 0.4 seconds. 
Starting ==> openrtm_aist                                                                                   
Finished <== openrtm_aist [ 21.3 seconds ]                                                                  
[build] Finished.                                                                                           
[build] Runtime: 22.2 seconds 
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ catkin build openrtm_aist
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
Found '204' packages in 0.4 seconds. 
Starting ==> openrtm_aist                                                                                   
Finished <== openrtm_aist [ 21.4 seconds ]                                                                  
[build] Finished.                                                                                           
[build] Runtime: 22.3 seconds 

```

garaemon, 2015-02-06:
openrtm_ais takes20 seconds at least, openhrp3 takes 12 seconds.
And hrpsys takes 1 minutes. (every package was compiled before)

```
$ catkin build hrpsys
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
Found '204' packages in 0.4 seconds. 
Starting ==> openrtm_aist                                                                                   
Finished <== openrtm_aist [ 21.3 seconds ]                                                                  
Starting ==> openhrp3                                                                                       
Finished <== openhrp3     [ 12.9 seconds ]                                                                  
Starting ==> hrpsys                                                                                         
Finished <== hrpsys       [ 1 minute and 1.9 seconds ]                                                      
[build] Finished.                                                                                           
[build] Runtime: 1 minute and 37.0 seconds 
```

k-okada, 2015-02-06:
The problem is that upstream package does not have correct dependency.
One thing I do not understand correctly is that , whether if there is
cleaner way not to run 'install' on second time.
for example, in hrpsys, it copies to CMAKE_INSTALL.... when we run `make
install` and it is normal to copy again
if you run `make install` again, or there are way not to run make install
in second time.

- turn off make document by default ->
https://github.com/fkanehiro/hrpsys-base/blob/master/CMakeLists.txt#L164,
need to change at .traivs for document genration (
https://github.com/fkanehiro/hrpsys-base/blob/master/.travis.yml#L27)
- turn off making document when you compiled from catkin,
- `touch install_doc` os something to prevent copy the documen twice.

garaemon, 2015-02-06:
> The problem is that upstream package does not have correct dependency.

Yeah, hrpsys recompiles a lot of binaries