start-jsk/rtmros_common #651: rtmlaunch.py fails because it cannot find openrtm_aist

garaemon, 2015-02-06:
with latest hrpsys/openrtm configuration (it means without ros package wrapper and built with `catkin build`), rtmlaunch.py tells following error:
```
Traceback (most recent call last):
  File "/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_tools/scripts/rtmlaunch.py", line 4, in <module>
    roslib.load_manifest('openrtm_tools')
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslib/launcher.py", line 62, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslib/launcher.py", line 98, in _generate_python_path
    packages = get_depends(pkg, rospack) 
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslib/launcher.py", line 51, in get_depends
    vals = rospack.get_depends(package, implicit=True)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 236, in get_depends
    s.update(self.get_depends(p, implicit))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 230, in get_depends
    names = [p.name for p in self.get_manifest(name).depends]
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 163, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 199, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: openrtm_aist
ROS path [0]=/opt/ros/hydro/share/ros
ROS path [1]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/geneus
ROS path [2]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/assimp_devel
ROS path [3]=/home/hrpuser/ros/hydro/src/multisense/multisense_description
ROS path [4]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_model_tools/euscollada
ROS path [5]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/rtmbuild
ROS path [6]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_tools
ROS path [7]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus
ROS path [8]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_pr2eus/pr2eus
ROS path [9]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools
ROS path [10]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge
ROS path [11]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials
ROS path [12]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_ros_bridge
ROS path [13]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg-unreleased/jsk_robot/jsk_hrp2_ros_bridge
ROS path [14]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/jsk_footstep_controller
ROS path [15]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_ros_patch/image_view2
ROS path [16]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_coordination_system/dynamic_tf_publisher
ROS path [17]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_footstep_msgs
ROS path [18]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_gui_msgs
ROS path [19]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_hark_msgs
ROS path [20]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_msgs
ROS path [21]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/sklearn
ROS path [22]=/home/hrpuser/ros/hydro/src/view_controller_msgs
ROS path [23]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_topic_tools
ROS path [24]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros
ROS path [25]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_visualization/jsk_rviz_plugins
ROS path [26]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner
ROS path [27]=/opt/ros/hydro/share
ROS path [28]=/opt/ros/hydro/stacks
process[StabilizerServiceROSBridge-15]: started with pid [18099]
[rtmlaunch_hrpsys_ros_bridge-8] process has died [pid 17897, exit code 1, cmd /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_tools/scripts/rtmlaunch.py /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch USE_COMMON=true USE_WALKING=true USE_COLLISIONCHECK=false USE_IMPEDANCECONTROLLER=true USE_SOFTERRORLIMIT=false USE_IMAGESENSOR=false USE_ROBOTHARDWARE=true USE_GRASPCONTROLLER=false USE_SERVOCONTROLLER=false USE_TORQUECONTROLLER=false USE_TORQUEFILTER=false motor_states:=motor_states_org __name:=rtmlaunch_hrpsys_ros_bridge __log:=/home/hrpuser/.ros/log/901d71ec-ae0a-11e4-b0da-00031d0c7c5c/rtmlaunch_hrpsys_ros_bridge-8.log].
log file: /home/hrpuser/.ros/log/901d71ec-ae0a-11e4-b0da-00031d0c7c5c/rtmlaunch_hrpsys_ros_bridge-8*.log
```

Is there anybody who can ever successfully launch hrpsys_ros_bridge.launch?

garaemon, 2015-02-06:
hrpsys-config.test also fails

```
$ rostest hrpsys_tools test-hrpsys-config.test 
... logging to /home/hrpuser/.ros/log/rostest-hrp2017v-25128.log
Traceback (most recent call last):
  File "/opt/ros/hydro/bin/rostest", line 53, in <module>
    rostestmain()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rostest/__init__.py", line 268, in rostestmain
    _main()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rostest/rostest_main.py", line 142, in rostestmain
    testCase = rostest.runner.createUnitTest(pkg, test_file)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rostest/runner.py", line 227, in createUnitTest
    config = roslaunch.parent.load_config_default([test_file], _DEFAULT_TEST_PORT)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/config.py", line 428, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 698, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 670, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 638, in _recurse_load
    self._env_tag(tag, context, ros_config)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 491, in _env_tag
    self.load_env(context, ros_config, *self.reqd_attrs(tag, context, XmlLoader.ENV_ATTRS))
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 188, in reqd_attrs
    return [self.resolve_args(tag.attributes[a].value, context) for a in attrs]
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 159, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 299, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 312, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 137, in _find
    return _find_executable(resolve_without_path, a, [args[0], path], context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 173, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 199, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: openrtm_aist
ROS path [0]=/opt/ros/hydro/share/ros
ROS path [1]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/geneus
ROS path [2]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/assimp_devel
ROS path [3]=/home/hrpuser/ros/hydro/src/multisense/multisense_description
ROS path [4]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_model_tools/euscollada
ROS path [5]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/rtmbuild
ROS path [6]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_tools
ROS path [7]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus
ROS path [8]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_pr2eus/pr2eus
ROS path [9]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools
ROS path [10]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge
ROS path [11]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials
ROS path [12]=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_ros_bridge
ROS path [13]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg-unreleased/jsk_robot/jsk_hrp2_ros_bridge
ROS path [14]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/jsk_footstep_controller
ROS path [15]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_ros_patch/image_view2
ROS path [16]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_coordination_system/dynamic_tf_publisher
ROS path [17]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_footstep_msgs
ROS path [18]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_gui_msgs
ROS path [19]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_hark_msgs
ROS path [20]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_msgs
ROS path [21]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/sklearn
ROS path [22]=/home/hrpuser/ros/hydro/src/view_controller_msgs
ROS path [23]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_topic_tools
ROS path [24]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros
ROS path [25]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_visualization/jsk_rviz_plugins
ROS path [26]=/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner
ROS path [27]=/opt/ros/hydro/share
ROS path [28]=/opt/ros/hydro/stacks
hrpuser@hrp2017v:~/ros/hydro/src/rtm-ros-robotics/rtmros_common$ 

```

garaemon, 2015-02-06:
This small test code fails:
```xml
<launch>
  <node name="rtcd" pkg="openrtm_aist" type="rtcd" />
</launch>
```

garaemon, 2015-02-06:
we need to remove all the
1. `pkg="openrtm_aist"`
2. `$(find openrtm_aist)`
3. `pkg="hrpsys"`
4. `$(find hrpsys)`

I'd like to know other robots which uses hrpsys on indigo

k-okada, 2015-02-06:
please use deb version of openrtm/openhrp3 for now

garaemon, 2015-02-06:
how about hrpsys? use deb version?

✉︎ Ryohei Ueda

k-okada, 2015-02-06:
I think you're using latest version of hrpsys

garaemon, 2015-02-06:
I tried to use latest hrpsys wothout ros package wrapper, but it's not working because it's no longer ros package.
I revert the environment to ros-wrappers

✉︎ Ryohei Ueda

k-okada, 2015-02-06:
https://github.com/k-okada/jsk_common/commit/aaffd5b74f6bef8d671caefd85a4649c970d5d09 should fix this problem

k-okada, 2015-02-06:
also satisfy https://github.com/jsk-ros-pkg/jsk_common/issues/726 request

garaemon, 2015-02-06:
> k-okada/jsk_common@aaffd5b should fix this problem

Please check your code and merge after testing.

k-okada, 2015-02-06:
wait for -> http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/jsk.rosbuild/DEBIAN_REPOSITORY=ros,ROS_DISTRO=indigo,TARGET_PACKAGE=jsk-ros-pkg,USE_SOURCE=source,slave=master/583/console

garaemon, 2015-02-06:
> wait for -> http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/jsk.rosbuild/DEBIAN_REPOSITORY=ros,ROS_DISTRO=indigo,TARGET_PACKAGE=jsk-ros-pkg,USE_SOURCE=source,slave=master/583/console

Good! But the jenkins test is not enough. Please run test code.  
I'm really nervous and conservative about this topic now.

garaemon, 2015-02-06:
> Good! But the jenkins test is not enough. Please run test code.

At this momement, I think [test in hrpsys_tools](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_tools/test/test-hrpsys-config.test) and [test in hrpsys_ros_bridge](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/test/test-samplerobot.test) are good test code

k-okada, 2015-02-06:
if you still have non-catkin hrpsys environment, please check by adding
`export ROS_PACKAGE_PATH=`pwd`/devel/share/:$ROS_PACKAGE_PATH`
and see if this works

garaemon, 2015-02-06:
> if you still have non-catkin hrpsys environment, please check by adding
> `export ROS_PACKAGE_PATH=`pwd`/devel/share/:$ROS_PACKAGE_PATH`
> and see if this works

oops, I already removed it, however I also came up with this idea but forget to test it...

I will test it tomorrow.

k-okada, 2015-02-06:
that's what we do in
https://github.com/k-okada/jsk_common/blob/use_catkin_tools/jsk.rosbuild,
so it will ready on tomorrow.

130s, 2015-09-16:
If this still happens please reopen.

closed 2015-09-16 by 130s

k-okada, 2015-09-16:
please explain your setup and error code, also adding test program to
reproduce this would be fine.