start-jsk/rtmros_common #655: rtshell_setup.sh does not work with non-catkin rtshell

garaemon, 2015-02-10:
```
$ source rtshell-setup.sh 
[rospack] Error: stack/package rtshell not found
[rospack] Error: stack/package openrtm_aist_python not found
[rospack] Error: stack/package rtshell not found
[rospack] Error: stack/package rtshell not found
[rospack] Error: stack/package rtctree not found
[rospack] Error: stack/package rtsprofile not found
[rospack] Error: stack/package rtshell not found
Warning : Failed to load shell_support, try rosmake openrtm_tools
```

k-okada, 2015-02-10:
Warning : Failed to load shell_support, try rosmake openrtm_tools

 should be fixe via https://github.com/start-jsk/rtmros_common/pull/656,
but [rospack] Error: stack/package rtshell not found
 should not happen if you install via deb or use jsk_rosbuild containing
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk.rosbuild#L228

garaemon, 2015-02-10:
Hmm I set up with jsk.rosbuild.But it might fail because of too heavy github cloning

✉︎ Ryohei Ueda

https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk.rosbuild#L228