start-jsk/rtmros_common #669: start-impedance でエラーが出る。

YuOhara, 2015-02-25:
eus上でstart-impedanceすると以下のエラーが出ます。

```
 (send *ri* :start-impedance :arms :moment-gain (float-vector 0 0 0) :k-p 1000 :d-p 400)
/home/jskuser/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: multiple variable declaration in (apply #'send self :raw-set-impedance-controller-param (string-downcase name) args)
```

garaemon, 2015-02-25:
@snozawa hrpsysをadd_graspless_manip_modeにしないとダメですか?
`:raw-set-impedance-controller-param`の引数にnameという仮引数がダブってしまっています

snozawa, 2015-02-25:
問題はimpedanceなので関係ないと思っています。

ちなみにadd_graspless_manip_modeは今日mergeされた
https://github.com/fkanehiro/hrpsys-base/pull/498
に含まれていて、今EuslispのインターフェースもPRしました
https://github.com/start-jsk/rtmros_common/pull/670

@YuOhara さん
エラーがでた箇所で
`(btrace)`
`,name`
`,args`
などしてもらっていいでしょうか。

また、その環境での`roscd hrpsys_ros_bridge && git log | head -n1`を教えてください。

garaemon, 2015-02-25:
add_grasplessにしてコンパイルしたら動くようになってしまいました。
Impedanceのidlにnameという変数があるかないかが重要なポイントのようです

✉︎ Ryohei Ueda

fkanehiro/hrpsys-base#498

YuOhara, 2015-02-25:
エラーが出た箇所でコマンドを打ちました。
```
2.irteusgl$ send (send *ri* :start-impedance :arms :moment-gain (float-vector 0 0 0) :k-p 800 :d-p 400)
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: multiple variable declaration in (apply #'send self :raw-set-impedance-controller-param (string-downcase name) args)
3.E1-irteusgl$ 
nil
3.E1-irteusgl$ (btrace
)
euserror
euserror
(apply #'send self :raw-set-impedance-controller-param (string-downcase name) args)
(apply #'send self :raw-set-impedance-controller-param (string-downcase name) args)
(send* self :raw-set-impedance-controller-param (string-downcase name) args)
(apply method-func limb args)
(apply method-func limb args)
(cond ((eq limb :legs) (mapcar #'(lambda (l) (send* self method-name l args)) '(:rleg :lleg))) ((eq limb :arms) (mapcar #'(lambda (l) (send* self method-name l args)) '(:rarm :larm))) ((and (find-method robot limb) (car (send robot limb :force-sensors))) (apply method-func limb args)) (t (error ";; No such limb or force sensor is defined for ~A~%." limb)))
(apply #'send self :force-sensor-method limb #'(lambda (name &rest _args) (send* self :raw-set-impedance-controller-param (string-downcase name) args)) :set-impedance-controller-param args)
t
4.E1-irteusgl$ ,name
"rarm"
5.E1-irteusgl$ ,args
(:moment-gain #f(0.0 0.0 0.0) :k-p 800 :d-p 400)
```
```
$ roscd hrpsys_ros_bridge && git log | head -n 1
commit 4ad7dbf59551ed2897ce3411d38a0ac3cdb919ac
```

garaemon, 2015-02-25:
https://github.com/fkanehiro/hrpsys-base/commit/5f310dca5174af5be34cd8955259273165a9db52#diff-7100325148afb9fedc80b245672a9f49L17

この変更の後だと動く来ますね。これがないと、nameという引数が衝突する
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L441

snozawa, 2015-02-25:
そうですね。
rtm-ros-robot-interface.lをつかうマシンで最新hrpsysで動くと思います。

closed 2015-02-26 by garaemon