start-jsk/rtmros_common #691: collision_state fail to narrow 'CollisionDetectorService'

k-okada, 2015-04-01:
colliosin_state.py could not launch due to following errors
```
/home/k-okada/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/collision_state.py:177: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub_collision = rospy.Publisher('collision_detector_marker_array', MarkerArray)
configuration ORB with  localhost : 15005
Traceback (most recent call last):
  File "/home/k-okada/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/collision_state.py", line 181, in <module>
    rtc_init()
  File "/home/k-okada/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/collision_state.py", line 59, in rtc_init
    co_svc = narrow(co.service("service0"), "CollisionDetectorService")
  File "/home/k-okada/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/rtm.py", line 724, in narrow
    return ior._narrow(getattr(sys.modules[package], klass))
AttributeError: 'module' object has no attribute 'CollisionDetectorService'
```

I'm still not understand python's import behavior, but in this case OpenHRP point to `dist-packages/OpenHRP` and this only contains 
```
 # DO NOT EDIT THIS FILE!
#
# Python module OpenHRP generated by omniidl

import omniORB; from OpenRTM_aist.RTM_IDL import SDOPackage_idl
omniORB.updateModule("OpenHRP3")

# ** 1. Stub files contributing to this module
import PathPlanner_idl
import OpenHRPCommon_idl
import World_idl
import ModelLoader_idl
import CollisionDetector_idl
import DynamicsSimulator_idl
import Controller_idl
import ViewSimulator_idl
import OnlineViewer_idl
import ClockGenerator_idl

```
whereas `dist-packages/hrpsys/OpenHRP/__init__.py` includes `import CollisionDetectorService_idl`, but not sure if this is nominal settings, because it only happens for me.

workaround for this:

```
diff --git a/hrpsys_ros_bridge/scripts/collision_state.py b/hrpsys_ros_bridge/scripts/collision_state.py
index ec2bb13..fa9c8fb 100755
--- a/hrpsys_ros_bridge/scripts/collision_state.py
+++ b/hrpsys_ros_bridge/scripts/collision_state.py
@@ -14,7 +14,8 @@ import hrpsys
 import rtm
 
 from rtm import *
-from OpenHRP import *
+from hrpsys.OpenHRP import *
+import OpenHRP
 
 import socket
 import time
```

eisoku9618, 2015-04-01:
you must use latest one
https://github.com/start-jsk/rtmros_common/commit/3c53c66d183a3a59f4a3ef912a8e2baf0aa81545

k-okada, 2015-04-01:
exactly

closed 2015-04-01 by k-okada