start-jsk/rtmros_common #692: hrpsys_ros_bridgeを立ち上げるとCPUパワーが１００％になる

eisoku9618, 2015-04-02:
https://github.com/start-jsk/rtmros_common/issues/408
と同じですがなにかおかしいです．@k-okada

[image]

k-okada, 2015-04-04:
@eisoku9618  チョット教えて欲しいんだけど，`rosparam get use_sim_time` はさすがにfalseだよね．

時間があったら以下をfalseにしてCollisionDetectorServiceROSBridgeを立ち上げずactivateもconnectもしないと，どうなるか教えて欲しいんだけど，
注目点は/collision_state.py  のloadで，rosbrigeをはしらせずにrtcdだけの状態でこれを上げると多分CPU使用率は数％なんだけど，手元だとなぜか他の不可が上がるとこちらも増えてくるんだけど，実機も同じことが起こっているか知りたいところです．

```
diff --git a/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch b/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch
index 838dc03..2e123df 100644
--- a/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch
+++ b/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch
@@ -7,7 +7,7 @@
   <arg name="INSTALL_ROBOT_DESCRIPTION" default="true" />
   <arg name="DEBUG_HRPSYS" default="false" />
 
-  <arg name="USE_DIAGNOSTICS" default="true" />
+  <arg name="USE_DIAGNOSTICS" default="false" />
 
   <!-- Set these values false when using HIRO -->
   <arg name="USE_COMMON" default="true" />
@@ -21,7 +21,7 @@
   <arg name="USE_SOFTERRORLIMIT" default="true" />
   <arg name="USE_TORQUEFILTER" default="false" />
   <arg name="USE_IMAGESENSOR" default="false" />
-  <arg name="USE_HRPSYS_PROFILE" default="true" />
+  <arg name="USE_HRPSYS_PROFILE" default="false" />
```
```
@@ -143,11 +143,11 @@
        CollisionDetector
   -->
   <group if="$(arg USE_COLLISIONCHECK)" >
-    <node pkg="hrpsys_ros_bridge" name="CollisionDetectorServiceROSBridge" type="CollisionDetectorServiceROSBridgeComp"
+    <anode pkg="hrpsys_ros_bridge" name="CollisionDetectorServiceROSBridge" type="CollisionDetectorServiceROSBridgeComp"
           output="$(arg OUTPUT)" args="$(arg openrtm_args)" />
 
-    <rtconnect from="CollisionDetectorServiceROSBridge.rtc:CollisionDetectorService" to="co.rtc:CollisionDetectorService" subscription_type="new"/>
-    <rtactivate component="CollisionDetectorServiceROSBridge.rtc" />
+    <artconnect from="CollisionDetectorServiceROSBridge.rtc:CollisionDetectorService" to="co.rtc:CollisionDetectorService" subscription_type="new"/>
+    <artactivate component="CollisionDetectorServiceROSBridge.rtc" />
     <node pkg="hrpsys_ros_bridge" name="collision_state" type="collision_state.py"
           args="$(arg MODEL_FILE) $(arg omniorb_args)" output="$(arg OUTPUT)"/>
   </group>
```

eisoku9618, 2015-04-04:
> rosparam get use_sim_time はさすがにfalseだよね．

以下のように```use_sim_time```はないみたいです．

```
leus@jjiro:~$ rosparam list
/HrpsysSeqStateROSBridge/publish_sensor_transforms
/controller_configuration
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/computers/contains
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/computers/path
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/computers/type
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/hrpsys/contains
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/hrpsys/path
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/hrpsys/type
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/mode/contains
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/mode/path
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/mode/type
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/motor/contains
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/motor/path
/diagnostic_aggregator_jaxon_ros_bridge/analyzers/motor/type
/diagnostic_aggregator_jaxon_ros_bridge/base_path
/diagnostic_aggregator_jaxon_ros_bridge/pub_rate
/hrpsys_config_path
/hrpsys_config_script_name
/hrpsys_configurator_class_name
/hrpsys_nshost
/hrpsys_ros_bridge_start_command
/hrpsys_ros_bridge_stop_command
/hrpsys_rtcd_start_command
/hrpsys_rtcd_stop_command
/hrpsys_servo_start_command
/hrpsys_servo_stop_command
/robot_description
/robot_pose_ekf/debug
/robot_pose_ekf/freq
/robot_pose_ekf/imu_used
/robot_pose_ekf/odom_used
/robot_pose_ekf/output_frame
/robot_pose_ekf/self_diagnose
/robot_pose_ekf/sensor_timeout
/robot_pose_ekf/vo_used
/rosdistro
/roslaunch/uris/host_192_168_96_130__47911
/roslaunch/uris/host_192_168_96_130__48201
/roslaunch/uris/host_192_168_96_131__38292
/roslaunch/uris/host_192_168_97_158__41372
/roslaunch/uris/host_192_168_97_48__41793
/roslaunch/uris/host_192_168_97_48__45043
/roslaunch/uris/host_192_168_97_48__46201
/roslaunch/uris/host_192_168_97_48__46626
/roslaunch/uris/host_192_168_97_48__50232
/roslaunch/uris/host_192_168_97_48__56275
/roslaunch/uris/host_192_168_97_48__59760
/roslaunch/uris/host_ocs3__43518
/roslaunch/uris/host_transvm__44902
/rosversion
/run_id
/thk_hand_controller/left_hand_id
/thk_hand_controller/right_hand_id
```

k-okada, 2015-04-18:
c.f. http://answers.ros.org/question/29425/rospy-horrible-performance-with-sim_time/