start-jsk/rtmros_common #758: Release Request of 1.2.14

garaemon, 2015-06-23:
I'd like to use code include https://github.com/start-jsk/rtmros_common/commit/f18094ce39c4e5295bb39292444653083cb1bf05 , is it ok to release new version?

k-okada, 2015-06-23:
> I'd like to use code include f18094c
> , is it ok to release new version?

done

closed 2015-06-23 by garaemon

mmurooka, 2015-06-24:
今となってはそこまで重要でないのでメモ程度ですが，

https://github.com/start-jsk/rtmros_common/pull/744#issuecomment-113512907
でリリースをお願いしてしていただいて，

tork-a/rtmros_common-release の方は，
https://github.com/tork-a/rtmros_common-release#rtmros_common-hydro---1213-0 は
```
rtmros_common (hydro) - 1.2.13-0
The packages in the rtmros_common repository were released into the hydro distro by running /var/lib/jenkins/send_to_bloom.py start-jsk rtmros_common on Fri, 19 Jun 2015 09:18:46 -0000
```
のように [1]  より後になされていますが，

start-jsk/rtmros_common の方は，
https://github.com/start-jsk/rtmros_common/commit/cf846a8a83cfd80ad56eef168cb327e56aaa8b9f
で
```
 +1.2.13 (2015-06-11) 
```
のように[1]より前になっていました．

[1] https://github.com/start-jsk/rtmros_common/commit/f18094ce39c4e5295bb39292444653083cb1bf05 （今回必要なコミット）