start-jsk/rtmros_common #770: EmergencyStopperの状態をeusで取得したい

mmurooka, 2015-07-15:
@snozawaさん
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridge.cpp#L791-L797
でpublishされているemergency_stopperの状態を
eusで受け取りたいのですが，
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L1109-L1123
の下にこのトピックをsubscribeするメソッドを追加してもよろしいでしょうか．

mmurooka, 2015-07-15:
cc: @YutaKojio

k-okada, 2015-07-15:
もうそろそろopenrtmのポートを指定したら自動で読み込んでrosでpublishするプログラムができるといいですね．

mmurooka, 2015-07-15:
大変そうですが，そう思います．
もしやるとしたらポートに関する何かしらをIDLに書くようにするところから始めないといけない（というので合っていますか）

k-okada, 2015-07-15:
port名とその型を引数にして動くプログラムができると思います．，で，できるかな．

snozawa, 2015-07-15:
@mmurooka 
ポートをだすものをつくる場合は、@k-okadaさんのおっしゃるやり方がよいです。
IDLのように静的なファイルにかいてあるものだけでなく、例えばセンサのポートは
ModelLoaderのセンサ情報に基づいて動的につけたりしてるので、そういうのにも対応がいります。
ちなみに、
```
hcf.kf("acc")
hcf.kf("acc").get_port_profile()
hcf.kf("acc").get_port_profile().properties
```
みたいなのでポート情報が動的にえられます。

snozawa, 2015-07-15:
https://github.com/start-jsk/rtmros_common/issues/770#issue-95129274
は、roseusでsubscribeする場合は
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L13
このへんでコールバックをつくっています。

話がそれますが、メソッドの追加は、
`getEmergencyStopperParam`、
`setEmergencyStopperParam`
対応のめそっどがなかったので、追加してもらえるとたすかります。

https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L1094
のように追加したら、メソッドが追加されます。

k-okada, 2015-07-15:
引数ににしたプログラムにするのは静的動的かかわらず，指定したものを処理する，というスタイルですね．
でプログラムとしては限りなくopenrtmレイヤで書けるものになるかと思います．pythonで書いておいて，
必要だったらモジュールでインポートしてhcfかなにかから動的に，という感じだと負います．

snozawa, 2015-08-24:
@mmurooka さん
これOK？

mmurooka, 2015-08-24:
はい，https://github.com/start-jsk/rtmros_common/pull/772でOKになりましたので，closeします．

> もうそろそろopenrtmのポートを指定したら自動で読み込んでrosでpublishするプログラムができるといいですね．

は，別でIssue立てます．

closed 2015-08-24 by mmurooka