start-jsk/rtmros_common #774: HrpsysSeqStateROSBridgeでHRP3ハンドのトピックを出す

mmurooka, 2015-07-17:
@garaemonさん

HrpsysSeqStateRosBridge （https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridge.cpp）から
HRP3ハンドの関節角を出したいという話でしたが，
@snozawa さんとはなしていて，
全身の関節角とハンドの関節角をつなげるのはどこでやるのがよろしいでしょうか．

とりあえずハンド関節角はjoint_states_handsとかで出しておいて，
全身とくっつけるのは別プログラムでよろしいでしょうか．

garaemon, 2015-07-17:
> とりあえずハンド関節角はjoint_states_handsとかで出しておいて，
> 全身とくっつけるのは別プログラムでよろしいでしょうか．

どちらでもいいです．
別トピックで出したら外側でrelayします．

mmurooka, 2015-07-19:
https://github.com/start-jsk/rtmros_hrp2/pull/259, https://github.com/start-jsk/rtmros_hrp2/pull/260, https://github.com/start-jsk/rtmros_common/pull/776 で、ハンドの関節角が/hand_joint_statesとしてpublishされるようになりました。

mmurooka, 2015-07-23:
@k-okada 先生，@garaemon さん，@snozawa さん

rtcのポートからrosのトピックにするのは，
https://github.com/start-jsk/rtmros_common/pull/776#issuecomment-122649858 や https://github.com/start-jsk/rtmros_common/issues/770#issuecomment-121523997 で言っているようにやったらいいことだとはおもうのですが，

今回はハンドの関節角はrtcのポートから出ていますが，
関節名をもっていないので，rosのjoint_statesにするためには，何らかの方法で与える必要があるので，
ただのrtc-ros変換では済まない話のように思います．
SeqStateROSBridgeに書くにしては一般的ではなさすぎるということなので，
専用のノードをひとつ作るのでよろしいでしょうか．

また，関節名の与え方は，
1.HRP3ハンド用のプログラムと割りきってソースにベタ書きするか，2.paramで与えるか，
3.全身関節角と同様にSeqStateROSBridgeでwrlのロボットモデルをロードして関節名を取得するか，
どのようにするのがよろしいでしょうか．
ハンドのために3までやるのは大げさなような気もします．

ハンドのjoint_statesを出したい理由は，@garaemonと話していて，ポイントクラウドのセルフフィルターでハンドの点群も除去するのに使えるからです．

k-okada, 2015-07-23:
> #776 (comment)
> や #770 (comment)

体内からでるStateにハンドを含めるというのはないでしょうか．．ColisionCheckにハンドを含めたいとなるんじゃないかとおもっているんですが，その可能性はないですか．
そうでなければ，１だとおもいますが，それをやるなら，HandRTCから関節名を取ってこれるようにするというのも有ると思います．

k-okada, 2015-07-23:
自分が触れる／触りやすい選択をすることで全体の整合性が取れない不可解なものになってく，というのがあるんじゃないかというのが心配点かな

garaemon, 2015-07-23:
> <javascript:_e(%7B%7D,'cvml','k-okada@jsk.t.u-tokyo.ac.jp');>>:

そうなると、corbaにjoint
states型みたいなのを導入することになって、普段使っているq_inとかとの整合性が少し気持ち悪くはないでしょうか？

k-okada, 2015-07-23:
体内のstateholderにハンドをいれればいまとおなじ関節をんがし，ジョイント名はモデルから，になって，
ハンドのRTCからしゅとくするには，そもそもハンドRTCにどうやって指示していたか忘れたけど，どこかのパラメータにあるなら，それをとってきて，（とれない？），そうでなければ，サービスでいいのでは．

snozawa, 2015-07-23:
一つ確認したいのは、結局joint_statesをつなげるのはやらなくてよい？
https://github.com/start-jsk/rtmros_common/issues/774#issuecomment-122248066
これによって、Barretハンド、ROBOTIqはんど、THKハンドやもともとRTCでなくて動いてるハンドとHRP３ハンドを共通にできるようにするか、
そうでなくてRTCのハンドのやり方を実装してくことになるのかかわりそうです。

また他のハンド同様、VRMLレベルでは今はハンド情報は使っておらず、URDFなどでくっつけるようになってます。
ハンドも関節名はRTCレベルではもってません。

garaemon, 2015-07-23:
> 一つ確認したいのは、結局joint_statesをつなげるのはやらなくてよい？
> #774 (comment)

どちらでもいいです，何も考えずに`/joint_states`に入れれば基本的には動くはずです．

mmurooka, 2017-02-03:
cc @iory