start-jsk/rtmros_common #782: collision_state.py occupies 100% CPU

garaemon, 2015-07-29:
collision_state.py looks heavy but I believe it just read collision detector's sate.

[image]

garaemon, 2015-07-29:
50Hz -> 100%
10Hz -> 20%
1Hz -> 2%

OTH, HrpsysSeqStateROSBridge is 40% in 100Hz

k-okada, 2015-07-29:
please check other process, too. I think everynode is very heavy than we
expected.
https://github.com/start-jsk/rtmros_common/issues/408#issuecomment-40581853

garaemon, 2015-07-29:
[image]

The top python is collision_state.py

k-okada, 2015-07-29:
so you're using everything in same machine without using /clock, that's
different than my use case.

- if /clock is published rospy becomes very heavy (
http://answers.ros.org/question/29425/rospy-horrible-performance-with-sim_time/
)
- if rtcd and collision_state.py run in different machine,
collision_state.py is heavy but also othre nodes are heavy

Please check if CPU usage if rtcd decrease, when you kill
collision_state.py,

k-okada, 2015-07-29:
also I'd like to know all command line arguments when you started
collsion_state.py

garaemon, 2015-07-29:
> so you're using everything in same machine without using /clock, that's
> different than my use case.
Our situation is
* no /clock
* everything run in same machine

> Please check if CPU usage if rtcd decrease, when you kill
> collision_state.py,

* rtcd load slightly decreased when i killed collision_state.py but it's really small amount. (75 -> 73) It looks a noise.

> also I'd like to know all command line arguments when you started
> collsion_state.py

```
$ roslaunch jaxon_ros_bridge jaxon_ros_bridge.launch --args /collision_state
LANG=C SIMULATOR_NAME=RobotHardware0 ORBgiopMaxMsgSize=2147483648 /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/collision_state.py /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_hrp2/jsk_models/JAXON_RED/JAXON_REDmain.wrl -ORBInitRef NameService=corbaloc:iiop:localhost:15005/NameService __name:=collision_state

```

k-okada, 2015-07-30:
collisoin_state.py should output something like follows; I'd like to know
the number of collision state in your case.

```
[INFO] [WallTime: 1438238930.107318] check 4 collision status, 0.796812 Hz
```