start-jsk/rtmros_common #792: sequence<sequence<xxx>>のidlを変換できない

eisoku9618, 2015-08-18:
```lisp
(send *ri* :set-foot-steps
      (list (make-coords :coords (send *robot* :lleg :end-coords :copy-worldcoords) :name :lleg)
            (make-coords :coords (send (send *robot* :rleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :rleg)
            (make-coords :coords (send (send *robot* :lleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :lleg)))
```

のようにfootstepのlistを与えていた部分を

```lisp
(send *ri* :set-foot-steps
      (list (list (make-coords :coords (send *robot* :lleg :end-coords :copy-worldcoords) :name :lleg)
                  (make-coords :coords (send *robot* :lleg :end-coords :copy-worldcoords) :name :rarm))
            (list (make-coords :coords (send (send *robot* :rleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :rleg)
                  (make-coords :coords (send (send *robot* :rleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :larm))
            (list (make-coords :coords (send (send *robot* :lleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :lleg)
                  (make-coords :coords (send (send *robot* :lleg :end-coords :copy-worldcoords) :translate (float-vector 200 0 0)) :name :rarm))))
```

のようにfootstepのlistのlistを与えられるようにしたくて，hrpsys-baseのidlの

```cpp
    typedef sequence<Footstep> FootstepSequence;
    boolean setFootSteps(in FootstepSequence fs, in long overwrite_fs_idx);
```
のところに
```cpp
    typedef sequence< sequence<Footstep> > FootstepNodesSequence;
    boolean setFootStepNodes(in FootstepNodesSequence fns, in long overwrite_fs_idx);
```

https://github.com/eisoku9618/hrpsys-base/blob/add-interface-of-set-footstep/idl/AutoBalancerService.idl#L50-L51

を追加してみたのですが，hrpsys_ros_bridgeをビルドするときに，[``rtmbuild_genidl``](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/catkin.cmake#L91) の中で呼ばれる [rtmbuild_init](https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/cmake/rtmbuild.cmake#L36) の中で呼ばれる[_rtmbuild_genbridge_init](https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/cmake/rtmbuild.cmake#L113)の中で呼ばれる[ここ](https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/cmake/servicebridge.cmake#L38) で実行されている[idl2srv.py](https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/scripts/idl2srv.py)の[ここ](https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/scripts/idl2srv.py#L232)で落ちます．

エラーは以下になります．

```
AttributeError: Struct instance has no attribute 'kind'
```

https://gist.githubusercontent.com/eisoku9618/f7290d77283c2f72464d/raw/29d67d55478c80e61f5386e81ce5691cf0d6822d/a.txt

k-okada, 2015-08-19:
1) 出て欲しいpythonのコードはどうなるか教えて欲しいのと，
2) ` typedef sequence< <FootstepSequence> FootstepNodesSequence;`
とするとどうなるか教えて欲しいです．

eisoku9618, 2015-08-19:
ご返信ありがとうございます．

> 1) 出て欲しいpythonのコードはどうなるか教えて欲しいのと，

現状は``hrpsys_ros_bridge/msg/OpenHRP_AutoBalancerService_Footstep.msg``が

```cpp
float64[3] pos
float64[4] rot
string leg
```

で，``hrpsys_ros_bridge/srv/OpenHRP_AutoBalancerService_setFootSteps.srv``が，

```cpp
OpenHRP_AutoBalancerService_Footstep[] fs
int32 overwrite_fs_idx
---
bool operation_return
```

となっていまして，欲しいコードは

``hrpsys_ros_bridge/msg/OpenHRP_AutoBalancerService_FootstepNodes.msg``の

```cpp
OpenHRP_AutoBalancerService_Footstep[] fs
```

という新しいメッセージと

``hrpsys_ros_bridge/srv/OpenHRP_AutoBalancerService_setFootStepNodes.srv``の

```cpp
OpenHRP_AutoBalancerService_FootstepNodes[] fns
int32 overwrite_fs_idx
---
bool operation_return
```

という新しいサービスです．

> 2) ` typedef sequence< <FootstepSequence> FootstepNodesSequence;`

こちらに変更しても同じエラーになります．

eisoku9618, 2015-08-19:
自己解決した気がします．

idlに書かれているstructがmsgになって，idlに書かれている関数の引数がsrvになるのだという理解に達し，

```cpp
    struct FootstepNodes
    {
        sequence<Footstep> fs;
    };
```

というように新たにstructを加えればいいのだという事が分かりました．

```cpp
    typedef sequence< sequence<Footstep> > FootstepNodesSequence;
    boolean setFootStepNodes(in FootstepNodesSequence fns, in long overwrite_fs_idx);
```

ではなくて，

```cpp
    struct XXXXX
    {
        sequence<Footstep> fs;
    };
    typedef sequence<XXXXX> FootstepNodesSequence;
    boolean setFootStepNodes(in FootstepNodesSequence fns, in long overwrite_fs_idx);
```

のように一回structをかませることで，現状の変換コードで idl から所望の srv / msg が作られるようになりました．

ありがとうございました．

k-okada, 2015-08-19:
すごい，すごすぎるデバッグ力で尊敬するけど，だめではないかい．
https://github.com/start-jsk/rtmros_common/blob/master/rtmbuild/scripts/idl2srv.py#L204
のどこかを直して対応したいところ．

eisoku9618, 2015-08-20:
``sequence <sequence <xxx> >`` をしたいときは

```cpp
    struct yyy
    {
        sequence<xxx> hoge;
    };
    typedef sequence<yyy> fuga;
```

のように一回``yyy``をかませることで解決できる，というのではダメで，

```cpp
    typedef sequence< sequence<xxx> > fuga;
```

ができて然るべきだ，ということでしょうか？

snozawa, 2015-08-20:
そうですね。
基本的にはコンバータなので、idlで標準で記述できて、かつROSのmsg/srvでも問題ない記法であれば変換できるようにするのがベターですね。

k-okada, 2015-08-20:
ダメじゃないけどダサいよね．

eisoku9618, 2015-08-20:
分かりました．ひとまずreopenしておきます．