start-jsk/rtmros_common #813: 3rd test takes too much time exceeding 50 minutes

garaemon, 2015-09-04:
https://github.com/start-jsk/rtmros_common/blob/master/.travis.yml#L23
https://travis-ci.org/start-jsk/rtmros_common/jobs/78539507

Can we remove it?

It compiles openhrp3 and hrpsys

garaemon, 2015-09-04:
cc: @orikuma

garaemon, 2015-09-04:
Or use jenkins?

garaemon, 2015-09-04:
* https://github.com/start-jsk/rtmros_common/pull/812
* https://github.com/start-jsk/rtmros_common/pull/811
* https://github.com/start-jsk/rtmros_common/pull/804

These  3 PRs failed to compile the 3rd test.

But I think this test is important, with lthe atest openhrp3 and hrpsys.

garaemon, 2015-09-04:
It specifies NOT_TEST_INSTALL="true" but it looks it runs https://github.com/jsk-ros-pkg/jsk_travis/blob/master/travis.sh#L172

garaemon, 2015-09-04:
> It specifies NOT_TEST_INSTALL="true" but it looks it runs https://github.com/jsk-ros-pkg/jsk_travis/blob/master/travis.sh#L172

it's not correct. The test ends during run_tests.

garaemon, 2015-09-04:
It may be slow because of `ROS_PARALLEL_JOBS="-j1 -l1"`

snozawa, 2015-09-04:
Third test is for euslisp test.
For euslisp test, I used dependent software from source.

However, we just need to use hrpsys-base from source, but openhrp3 can be deb.

snozawa, 2015-09-04:
https://github.com/start-jsk/rtmros_common/pull/806

snozawa, 2015-09-17:
If this problem occurs again, please reopen this issue.

closed 2015-09-17 by snozawa