start-jsk/rtmros_common #844: roseus samplerobot-interface.l が立ち上がらない

yoshimalucky, 2015-10-22:
roseus samplerobot-interface.l を実行すると以下のエラーが出ます。
ERROR th=0 unbound variable hrpsys_ros_bridge::openhrp_i\
mpedancecontrollerservice_objectturnarounddetectorparamhrpsys_ros_bridge2::Openhrp_impedancecontrollerservice_objectturnarounddetectorpara\
m in (eval param-class)E:

rtm-ros-robot-interface.l の中で以下のような関数があり、
(def-set-get-param-method 'hrpsys_ros_bridge::OpenHRP_ImpedanceControllerService_objectTurnaroundDetectorParam
  :set-object-turnaround-detector-param :get-object-turnaround-detector-param
  :impedancecontrollerservice_setObjectTurnaroundDetectorParam
  :impedancecontrollerservice_getObjectTurnaroundDetectorParam)

ここで呼ばれている
hrpsys_ros_bridge::OpenHRP_ImpedanceControllerService_objectTurnaroundDetectorParam
があるかどうかを確認したところ、
leus@artichoke:~/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge$ ls msg | grep Imp
OpenHRP_ImpedanceControllerService.msg
OpenHRP_ImpedanceControllerService_ControllerMode.msg
OpenHRP_ImpedanceControllerService_impedanceParam.msg

このように見当たらないようなので、msgが上手く生成できていないようです。
rm ~/ros/indigo/build
rm ~/ros/indigo/devel
をして、やり直すしかないでしょうか。

以下、全ログ：
leus@artichoke:~/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/euslisp$ roseus samplerobot-interface.l 
configuring by "/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtra\
n ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses\
 ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub \
;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xto\
p ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;\
; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; f\
stringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql 
;; extending gcstack 0x5ca32e0[16374] --> 0x5fa4030[32748] top=395f
irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb i\
rtgl irtviewer                                                                                                                             
EusLisp 9.15(4574ef7 1.0.10) for Linux64 created on artichoke(Thu Oct 22 13:44:45 JST 2015)
roseus ;; loading roseus("c26f61a") on euslisp((9.15 artichoke Thu Oct 22 13:44:45 JST 2015 4574ef7 1.0.10))                               
eustf roseus_c_util 
;; extending gcstack 0x5fa4030[32738] --> 0x71bafd0[65476] top=7ef2
/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 unbound variable hrpsys_ros_bridge::openhrp_i\
mpedancecontrollerservice_objectturnarounddetectorparamhrpsys_ros_bridge2::Openhrp_impedancecontrollerservice_objectturnarounddetectorpara\
m in (eval param-class)E:

snozawa, 2015-10-22:
```
cd ~/ros/indigo_parent/src/hrpsys
git pull
catkin bt
roscd hrpsys_ros_bridge
catkin bt --start-with-this --force-cmake
```
してみてください。

yoshimalucky, 2015-10-23:
ありがとうございます。

一応、それでやってみましたが、できていなくて、
おそらくですが、ros-indigo-hrpsysがapt-getで入っていると
もしかしたら上手くいかないのかもしれないので、
aptのパッケージを極力消してやってみます。

==================================
東京大学 情報理工学系研究科
創造情報学専攻 情報システム工学研究室
田中 義丸

〒113-8656 東京都文京区本郷7-3-1
工学部二号館 七階73A1～73B2
電話番号 03-5841-7416 FAX 03-3818-0835
==================================

snozawa, 2016-04-08:
おそらく上記で解決すると思うので閉じます。
必要になったらopenしてください。

closed 2016-04-08 by snozawa

k-okada, 2016-04-20:
apt で入っている状況でもtravisは通っていると思う．テストが足りない？
https://github.com/start-jsk/rtmros_common/pull/929

snozawa, 2016-04-20:
Euslispのテストは
https://github.com/start-jsk/rtmros_common/blob/master/.travis.yml#L26
の箇所でIS_EUSLISP_TRAVIS_TEST="true"にしたときに走るようになります。

ちなみに、#929のテストは
- hrpsysがsourceかdebか
- rosbridgeがsourceかdebか

のどの組み合わせのテストでしょうか？

k-okada, 2016-04-20:
#929 ですが，rosdbridgeをdebでいれているので，hrpsys, rosbridge ともに debが入っている．
で，かつ，
rosbridgeをソースで入れる
rosbridge + hrpsys+ openhrp3 をソースで入れる，
状態でコンパイルしているつもりです．

snozawa, 2016-04-20:
なるほど、わかりました。
おそらくその状態で
`IS_EUSLISP_TRAVIS_TEST="true"`
にするとこのissueのテストもされるようになると思います。