start-jsk/rtmros_common #859: ImageSensorROSBridgeでカメラパラメタのDに値を入れる

YuOhara, 2015-11-13:
choreonoidを使っているのですが、
カメラインフォのD情報を入れたいと思っています。
```
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/src$ git diff ImageSensorROSBridge.cpp 
diff --git a/hrpsys_ros_bridge/src/ImageSensorROSBridge.cpp b/hrpsys_ros_bridge/
index 3a1c956..a68e973 100644
--- a/hrpsys_ros_bridge/src/ImageSensorROSBridge.cpp
+++ b/hrpsys_ros_bridge/src/ImageSensorROSBridge.cpp
@@ -206,6 +206,7 @@ RTC::ReturnCode_t ImageSensorROSBridge::onExecute(RTC::Uniqu
     info->width  = image->width;
     info->height = image->height;
     info->distortion_model = "plumb_bob";
+    info->D.resize(5);
     info->K = K;
     info->P = P;
     info->R[0] = info->R[4] = info->R[8] = 1;
```
という方法しかわからなかったのですが、正しくはどうするべきなのでしょうか？

pinholeカメラモデルを使いたくて、Dが空だとプログラムが死んでしまいます。

YoheiKakiuchi, 2015-11-13:
現状はDが空だったかな？

> という方法しかわからなかったのですが、正しくはどうするべきなのでしょうか？

これは、どういう意味かな？
これでプログラムが動けば、問題ないんじゃないかと思います。

YuOhara, 2015-11-14:
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/ImageSensorROSBridge.cpp#L264

ここのif分の中で、resizeしているコードあるのでこのifに入れたらよいと思うのですが、
条件がよく理解できていないです。

YuOhara, 2015-11-14:
> 現状はDが空だったかな？

Dは空でした。

k-okada, 2015-11-15:
hrpsysだかchorenooid側からデータが来ていないのでは？

YoheiKakiuchi, 2015-11-16:
> hrpsysだかchorenooid側からデータが来ていないのでは？

そうですね、choreonoidではDのパラメータを入れていないですね。

> https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/ImageSensorROSBridge.cpp#L264

ここは、RTMのメッセージにパラメータが入ってきたら、それを使うようになっていて、
パラメータが入って来ていないときは何もしていないので、空になってしまいますね。

YuOhara, 2015-11-16:
かぶってしまいましたが、ちょうどchoreonoidで値を入れてないのを確認していたところでした。
```
leus@leus-ThinkPad-W541:choreonoid [master]$ grep -r "matrix_element" *
src/OpenRTMPlugin/corba/CameraImage.idl:  double matrix_element[5];
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.matrix_element[0] = fu;
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.matrix_element[1] = 0.0;
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.matrix_element[2] = u0;
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.matrix_element[3] = fv;
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.matrix_element[4] = v0;
leus@leus-ThinkPad-W541:choreonoid [master]$ grep -r "distortion_" *
src/OpenRTMPlugin/corba/CameraImage.idl:  sequence<double> distortion_coefficient;
src/OpenRTMPlugin/VirtualRobotPortHandler.cpp:        value.data.intrinsic.distortion_coefficient.length(0);
```

k-okada, 2015-11-16:
rtmros_commonを直すのではいいことだけど，それと同時に，

pinholeカメラモデルを使いたくて、Dが空だとプログラムが死んでしまいます。

こちらのプログラムを直しましょう．

YuOhara, 2015-11-19:
pinholeカメラの方にPRを送って、空のDでもプログラムが動くようになりました。

opencv的には、Dが空でも受け入れているようです。

k-okada, 2015-11-20:
すばらしい．で，ダブルチェックでrtmros_common側も直しておきましょう．

k-okada, 2015-11-20:
小原くんのナイスPR→　https://github.com/ros-perception/vision_opencv/pull/82
ログ見ると結構なJSK率ですね．素晴らしい．

YuOhara, 2015-11-20:
2つPR出しました。
A: choreonoid側の変更
https://github.com/s-nakaoka/choreonoid/pull/56

B: rtmros_commonの変更
https://github.com/start-jsk/rtmros_common/issues/859#issuecomment-156593708

どちらのパターンも最終的には同じ出力です。(カメラパラメタのDに[0.0, 0.0, 0.0, 0.0, 0.0]が入る)
もちろん両方採用されても大丈夫です。両方却下されても一応大丈夫です。
ただ、個人的にはAだけで十分だと思っていて、それは、
* 再上流で値を入れたほうがわかりやすい
* そのまま出した方が「bridge」という名前に即している

と考えるからです。

YuOhara, 2016-01-07:
解決しました。

closed 2016-01-07 by YuOhara