start-jsk/rtmros_common #861: 別のidlをincludeしているidlのbuildがうまく行かない

eisoku9618, 2015-11-18:
https://github.com/eisoku9618/hrpsys-base/commit/448e0effe0bf30271b5b22d93ed0fe596ef8b6fb のように、
hrpsys-baseのStabilizerService.idlの中でAutoBalancerService.idlをincludeした場合に、
hrpsys-baseのbuildは通るのですが、hrpsys_ros_bridgeのbuildがうまく行かないようです。

以下のエラーが出ます。

```bash
In file included from /home/leus/ros/hydro/devel/include/hrpsys_ros_bridge/idl/StabilizerServiceStub.h:30:0,
                 from /home/leus/ros/hydro/devel/include/hrpsys_ros_bridge/idl/StabilizerServiceStub.cpp:12:
/home/leus/ros/hydro/devel/include/hrpsys_ros_bridge/idl/StabilizerService.hh:23:34: fatal error: AutoBalancerService.hh: No such file or directory
compilation terminated.
```

eisoku9618, 2015-11-19:
自動生成されるファイル（StabilizerService.hh）のincludeの囲み文字が勝手に変更されてしまうのが原因のようです．

自動生成される``~/ros/hydro/devel/include/hrpsys_ros_bridge/idl/StabilizerService.hh``に書かれている``#include <AutoBalancerService.hh>``がダメで，直接ファイルを編集して``#include "AutoBalancerService.hh"``にしてカレントディレクトリを見るようにしたら通りました．

hrpsys-baseのbuildで作られる``StabilizerService.hh``では``#include "AutoBalancerService.hh"``になっているのに対し，
hrpsys_ros_bridgeのbuildで作られる``StabilizerService.hh``では``#include <AutoBalancerService.hh>``になっています．

eisoku9618, 2015-11-19:
XXXservice.hhは

https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/catkin.cmake#L91

で呼んでいる``rtmbuild_genidl``の中で作られていて，具体的には以下のコマンド

```bash
`pkg-config openrtm-aist  --variable=rtm_idlc` `pkg-config openrtm-aist  --variable=rtm_idlflags` -C/tmp/ `rospack find hrpsys_ros_bridge`/idl/StabilizerService.idl
```

| pkg-config openrtm-aist --variable=rtm_idlc | pkg-config openrtm-aist --variable=rtm_idlflags |
| --- | --- |
| /usr/bin/omniidl  | -bcxx -Wba -nf -I/home/eisoku/ros/hydro_parent/devel/include/openrtm-1.1/rtm/idl |

eisoku9618, 2015-11-19:
omniidlに以下のオプションが使えそう

```bash
-Wbuse_quotes  Use quotes in #include directives (e.g. "foo" rather than <foo>).
```

mmurooka, 2015-11-19:
idlをincludeするのが同じクラスを使いたいみたいな理由でしたら，
それぞれのidlに同じ定義を書いておくのでいいと思います
（と@snozawa さんがオフラインでコメントしていました．）

eisoku9618, 2015-11-19:
https://github.com/tork-a/openrtm_aist-release/pull/4 で解決しました．

- hrpsysの方ではCMakeLists.txtにて ``-Wbuse_quotes``を使うように書いている
   - https://github.com/fkanehiro/hrpsys-base/blob/master/idl/CMakeLists.txt#L54
- hrpsys_ros_bridgeの方ではCMakeLists.txtにて``pkg-config openrtm-aist --variable=rtm_idlflags``を使うように書いてある

となっていましたので，openrtm-aist.pcに``-Wbuse_quotes``を付け加えました．

closed 2015-11-19 by eisoku9618