start-jsk/rtmros_common #877: hztest sometimes fails

snozawa, 2015-12-15:
Currently, hztest of `test-samplerobot.test` sometimes fails.

In upstream `hrpsys-base` travis test,hztest sometimes fails:
https://api.travis-ci.org/jobs/96893172/log.txt?deansi=true
```
Done waiting, validating results
FAIL

======================================================================
FAIL: test_hz (__main__.HzTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/ros/hydro/share/rostest/nodes/hztest/hztest", line 115, in test_hz
    self._test_hz(hz, hzerror, topic, test_duration, wait_time)
  File "/opt/ros/hydro/share/rostest/nodes/hztest/hztest", line 185, in _test_hz
    (rate, self.min_rate))
AssertionError: average rate (317.757Hz) exceeded minimum (450.000Hz)

----------------------------------------------------------------------
Ran 1 test in 5.890s

FAILED (failures=1)
-------------------------------------------------------------
SUMMARY:
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0 []
 * FAILURES: 1 [test_hz]
```

garaemon, 2015-12-15:
Let's increase retry count
https://github.com/jsk-ros-pkg/jsk_common/pull/1303

snozawa, 2015-12-15:
Current retry count is 4, a little bit large number:
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/test/test-samplerobot.test#L17

k-okada, 2015-12-15:
super dirty workaround is to increase hzerror, if this test is to check if
odom is running at 500hz, not 200hz, nor 100hz, then set hzerror to 200 is
ok, (but if someone propose this, I would be very reluctant  ;-0)
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/test/test-samplerobot.test#L15

snozawa, 2015-12-15:
BTW, why this test program checks `/odom` topic instead of odom tf even if the test name is `hztest_tf`?

How about checking `/joint_states`?
`joint_states` relates with `tf`.

It is processed within same `if` sentence as hz output of ROS_INFO debug message 
and we can use ROS_INFO hz output for hint of hztest failure:
```
[ INFO] [1450170291.128860233, 14.552000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 100[Hz] (exec 356 Hz, dropped 0)
```
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridge.cpp#L441

k-okada, 2015-12-16:
On Tue, Dec 15, 2015 at 5:59 PM, Shunichi Nozawa <notifications@github.com>
wrote:

> BTW, why this test program checks /odom topic instead of odom tf even if
> the test name is hztest_tf?
>
>
may be I just copied them from somewhere, please fix them

--
◉ Kei Okada

snozawa, 2015-12-21:
> may be I just copied them from somewhere, please fix them

I changed `odom` => `joint_states`.
https://github.com/start-jsk/rtmros_common/pull/880

I tried some debug :
https://github.com/start-jsk/rtmros_common/pull/879#issuecomment-165090121
,but currently I could not found the reason of failure.

So I fixed this problem throguh
https://github.com/start-jsk/rtmros_common/issues/877#issuecomment-164669534
https://github.com/start-jsk/rtmros_common/pull/879

closed 2015-12-21 by snozawa