start-jsk/rtmros_common #878: Building hrpsys_ros_bridge is failed when genjava is installed

orikuma, 2015-12-15:
* Problem
In ubuntu 14.04 with indigo, hrpsys_ros_bridge failed to be built when genjava is installed.
ros-indigo-genjava is depended by ros-indigo-rosjava-messages which is depended by jsk packages such as jsk_android_apps or jsk_rosjava_messages (error messages are as follows).

* How to reproduce
```
roscd hrpsys_ros_bridge && git clean -xfd
catkin clean -a
sudo apt-get install ros-indigo-genjava
catkin build hrpsys_ros_bridge
```

* Appendix
The same error occurs in openrtm_ros_bridge in same situation.
This error doesn't occur and build is succedded when genjava does not installed.
Does ros_bridge supports java interface?

* Error log

```
[ 14%] [ 14%] Generating EusLisp code from hrpsys_ros_bridge/RTC_ModeCapable.msg
Generating Lisp code from hrpsys_ros_bridge/OpenHRP_WavPlayerService.msg
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
Traceback (most recent call last):
  File "/opt/ros/indigo/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s on the ros package path" % msg_pkg_name)
IOError: could not find hrpsys_ros_bridge on the ros package path
make[2]: *** [java/hrpsys_ros_bridge/build.gradle]  1
make[1]: *** [CMakeFiles/hrpsys_ros_bridge_generate_messages_java_gradle.dir/all]  2
make[1]: *** ....
[ 15%] Generating Lisp code from hrpsys_ros_bridge/OpenHRP_TorqueControllerService_torqueControllerParam.msg
[ 15%] [ 15%] Generating Lisp code from hrpsys_ros_bridge/RTC_DataFlowComponent.msg
Generating EusLisp code from hrpsys_ros_bridge/OpenHRP_StateHolderService.msg
[ 15%] [ 15%] Generating Lisp code from hrpsys_ros_bridge/OpenHRP_ImpedanceControllerService.msg
Generating EusLisp code from hrpsys_ros_bridge/OpenHRP_TimeKeeperService.msg
[ 15%] Generating EusLisp code from hrpsys_ros_bridge/RTC_PortService.msg
[ 15%] [ 15%] Generating Lisp code from hrpsys_ros_bridge/OpenHRP_ExecutionProfileService.msg
Generating EusLisp code from hrpsys_ros_bridge/OpenHRP_RobotHardwareService_RobotState.msg
[ 15%] WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
Traceback (most recent call last):
  File "/opt/ros/indigo/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s on the ros package path" % msg_pkg_name)
IOError: could not find hrpsys_ros_bridge on the ros package path
make[2]: *** [java/hrpsys_ros_bridge/build.gradle]  1
make[1]: *** [CMakeFiles/hrpsys_ros_bridge_generate_messages_java.dir/all]  2
[ 15%] Generating Lisp code from hrpsys_ros_bridge/OpenHRP_ExecutionProfileService_ComponentProfile.msg
Generating EusLisp code from hrpsys_ros_bridge/RTC_LightweightRTObject.msg
```

YuOhara, 2016-01-14:
same issue?
https://github.com/jsk-ros-pkg/jsk_visualization/issues/551

yoshimalucky, 2016-01-15:
The same error ossurs in pr2_2dnav and pr2_2dnav_slam which are not in our maintenance. And, these two are also fixed by adding message_generation in build_depend tag in package.xml.

YuOhara, 2016-01-15:
I think https://github.com/start-jsk/rtmros_common/issues/907 has fixed this.
If it is ok, please close.
(Off course it is possible that other pkgs have the same error)

YuOhara, 2016-01-15:
And we had better send pr to pr2_2dnav

k-okada, 2016-01-16:
sending PR to upstream is ok for now, but I think this is because genjava
did only support packages carefully designed, and fixing packages to
add/remove message_generation is not a good nor practical idea, so send PR
to genjava ->
https://github.com/rosjava/genjava/pull/4

P.S. Prof Ishiguro always ask us that "Is this fundamental problem?" and I
think this is true for software, does your PR solving fundamental problem?

wkentaro, 2016-01-16:
I think the problem is
* we do not see this tutorial when adding ros messages to a package
http://wiki.ros.org/msg so we forget listing message_generation in package.xml. (and does not `catkin p [package]` command so often, which automatically generates well configured CMakeLists.txt and pakcage.xml)
* catkin does not raise error for `add_message_files` or `add_service_files` even if `message_generation` and `message_runtime` is not listed in package.xml. (we notice that the package is not well configured with error: http://answers.ros.org/question/186986/cmake-error-unknown-cmake-command-add_message_files)

In my opinion, genjava should raise warning message when building.

k-okada, 2016-01-16:
#907 fixed this

closed 2016-01-16 by k-okada