start-jsk/rtmros_common #906: hrpsys-simulatorでnan/infがfoundと出る

YuOhara, 2016-01-13:
rtmlaunch hrpsys_ros_bridge_tutorials samplerobot.launch
して別ターミナルで、
(progn (load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l") (samplerobot-init) (send *ri* :angle-vector (send *sr* :reset-pose)) (send *ri* :wait-interpolation) (send *ri* :start-auto-balancer))
とすると
```
[abc,JointPathEx] ERROR nan/inf is found
```
とエラーが出て、そこからstを入れるシミュレータが落ちます（リスタートします）。

同じ理由で、choreonoidも最新のコードで落ちているように見えます。

ターミナルのログは以下です(hrpsys-simulator)
https://gist.github.com/YuOhara/b54ae20bae14ce222a43

snozawa, 2016-01-14:
hrpsys_ros_bridge_tutorials -> hrpsys_ros_bridgeにしてみて、
```
rtmlaunch hrpsys_ros_bridge samplerobot.launch USE_UNSTABLE_RTC:=true

(progn (load "package://hrpsys_ros_bridge/euslisp/samplerobot-interface.l") (samplerobot-init) (send *ri* :angle-vector (send *sr* :reset-pose)) (send *ri* :wait-interpolation) (send *ri* :start-auto-balancer))
```
ではどうでしょうか。

snozawa, 2016-01-14:
```
rtmlaunch hrpsys samplerobot.launch

python -i `rospack find hrpsys`/samples/SampleRobot/samplerobot_auto_balancer.py
```
の結果も教えてください。

snozawa, 2016-01-14:
また，
https://github.com/fkanehiro/hrpsys-base/pull/944
を試しにmergeしてみて結果がかわるかも教えてください．

YuOhara, 2016-01-14:
hrpsys->masterbranch はじめの出力 (rtm launch hrpsys_ros_bridge)
https://gist.github.com/YuOhara/f59e623927581df84b6e
hrpsys->masterbranch 二個目の出力　(rtmlaunch hrpsys)
https://gist.github.com/YuOhara/9b24d10b229fcd719323

YuOhara, 2016-01-14:
hrpsys->masterbranch はじめの出力 (rtm launch hrpsys_ros_bridge)
https://gist.github.com/YuOhara/48bbeea931917dd900bd
hrpsys->masterbranch 二個目の出力　(rtmlaunch hrpsys)
https://gist.github.com/YuOhara/a6313f2157f2481785d1

YuOhara, 2016-01-14:
結果は変わってないように見えて、hrpsys_ros_bridgeでabc入れるとエラーが発生してst入れるとsimulatorが落ちます

snozawa, 2016-01-14:
hrpsys-baseを
5652d17c93861a3ab11746af3c600f1f31dabf18
までもどしてみて，
hrpsys-base, hrpsys_ros_bridgeの両方を
```
catkin bt --start-with-this --force-cmake --cmake-args -DENABLE_DOXYGEN=OFF
```
してみてください．

YuOhara, 2016-01-17:
hrpsys->(5652d17c93861a3ab11746af3c600f1f31dabf18)branch はじめの出力 (rtm launch hrpsys_ros_bridge)

https://gist.github.com/YuOhara/e11ae8842c04d91d2d34
こちら、エラーなく動いているように見えます。

snozawa, 2016-01-25:
ところでこれはubuntu 14.04で、まだ解決してないでしょうか？

YuOhara, 2016-01-25:
14.04です。
解決しているかは、明日確認します

YuOhara, 2016-01-26:
14.04では解決していなかったです(rtm_launch hrpsys_ros_bridge samplerobot.launchをして、start-abc->start-stでhrpsys_simulatorが落ちました)
12.04マシンが手元にないので、そちらは確認できていないです。

snozawa, 2016-01-26:
12.04以前はtravis手元実機含め落ちてないですね。

snozawa, 2016-01-26:
マージされたので確認をお願いします

YuOhara, 2016-01-26:
hrpsys_simulator, choreonoidともにちゃんとabc, stが入ることを確認しました

closed 2016-01-26 by YuOhara