start-jsk/rtmros_common #919: チュートリアル通りにやってもST/ABCが起動しない

ishiguroJSK, 2016-04-07:
http://wiki.ros.org/rtmros_common/Tutorials/WorkingWithEusLisp#AutoBalancer_example
を見て勉強しているのですが、eusのコードを動かすと
```
Start and Stop Stabilizer => OK"
[ERROR] [1460018484.322483914, 67.512000000]: attempted to call service  /StabilizerServiceROSBridge/getParameter, but failed 
[ERROR] [1460018484.323956952, 67.512000000]: attempted to call service  /StabilizerServiceROSBridge/getParameter, but failed 
[ERROR] [1460018484.325871004, 67.516000000]: attempted to call service  /StabilizerServiceROSBridge/setParameter, but failed 
[ERROR] [1460018484.326638880, 67.516000000]: attempted to call service  /StabilizerServiceROSBridge/startStabilizer, but failed 
[ERROR] [1460018484.327525879, 67.518000000]: attempted to call service  /AutoBalancerServiceROSBridge/goPos, but failed 
[ERROR] [1460018484.328392919, 67.520000000]: attempted to call service  /AutoBalancerServiceROSBridge/waitFootSteps, but failed 
[ERROR] [1460018484.329130937, 67.520000000]: attempted to call service  /StabilizerServiceROSBridge/stopStabilizer, but failed 
```
とエラーが出るのですが、これは
```
rtmlaunch hrpsys_ros_bridge samplerobot.launch
```
ではなくて
```
rtmlaunch hrpsys_ros_bridge samplerobot.launch USE_UNSTABLE_RTC:=true
```
とするのが正しいでしょうか？

たしかにこれで起動してからeusのプログラムを走らせると`attempted to call service`のエラーは出ないのですが、今度はロボットが転倒してしまいます

mmurooka, 2016-04-07:
> rtmlaunch hrpsys_ros_bridge samplerobot.launch USE_UNSTABLE_RTC:=true
とするのが正しいでしょうか？

はい，そうなります．wikiの該当する部分の加筆・修正をお願いします．

stのような力フィードバックが必要な場合には，hrpsys-simulatorをトルク制御モードで起動する必要があります．（デフォルトはハイゲインモードになっています．）
例えば
http://wiki.ros.org/rtmros_common/Tutorials/AddRobotsEnvironmentsHrpsysSimulator#SampleRobot_example_3_:_Use_torque-based_simulation 
のようにすればできるはずです．

ishiguroJSK, 2016-04-07:
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/CMakeLists.txt#L411
```
generate_default_launch_eusinterface_files("$(find openhrp3)/share/OpenHRP-3.1/sample/model/sample1.wrl" hrpsys_ros_bridge_tutorials SampleRobot "--use-robot-hrpsys-config")
```
となっていて、
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L134
は
```
generate_default_launch_eusinterface_files(${_OPENHRP3_MODEL_DIR}/sample1.wrl hrpsys_ros_bridge SampleRobot)
```
です。
`hrpsys_ros_bridge_tutorials`のほうはhttps://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/cmake/compile_robot_model.cmake#L207
をみて、デフォルトの`USE_UNSTABLE_RTC`がtrueになっているので、tutorialのlaunchを上げるのが正解なのでしょうか。

snozawa, 2016-04-07:
>  をみて、デフォルトのUSE_UNSTABLE_RTCがtrueになっているので、tutorialのlaunchを上げるのが正解なのでしょうか。

hrpsys_ros_bridge_tutorialsは実際に研究室で使っている各ロボットの設定やサンプルがかいてあります（VRMLは別なところで管理）。
hrpsys_ors_bridgeはそれを含まない全ロボット共通なものと一部サンプルがはいっています
SampleRObotは、openhrp3(hrpsys_ros_bridgeのupstreamであるhrpsysのさらにupstream)本家にVRMLおいてある標準的なロボットなので、hrpsys_ros_bridgeでもサンプルをおいています。

SampleRobotのサンプルはhrpsys_ros_bridge_tutorialsから将来的にはけしたいところですが、
hrpsys_ros_bridge_tutorials中のSampleRobotのurdfファイルなどの変換をたくみにしている箇所がまだhrpsys_ros_bridgeレベルで解決できてません。

ということで、SampleRobotのサンプルに関しては特に問題なければhrpsys_ros_bridgeの方をつかってください。
ROS wikiも編集をしてもらえると助かります（編集は同じ部屋の人にきくと教えてくれると思います）

ishiguroJSK, 2016-04-08:
お二人とも回答有難うございます．
openhrp-project-generator ○○.xmlを使ってトルクシミュレーションモードでST/ABCを有効にして段差昇降確認できました．
ROSwikiの
http://wiki.ros.org/rtmros_common/Tutorials/WorkingWithEusLisp
も治せる箇所は直しておいたのですが，後半のチュートリアルの○○.xmlはハイゲインモードのものが同梱されているので，
現状，チュートリアルをやる各人が
```
openhrp-project-generator `rospack find openhrp3`/share/OpenHRP-3.1/sample/model/sample1.wrl `rospack find hrpsys`/share/hrpsys/samples/environments/TerrainFloor.wrl,-4.15,-0.2,0,1,0,0,0 --use-highgain-mode false --output `rospack find hrpsys`/share/hrpsys/samples/SampleRobot/MySetting.xml --timeStep 0.001 --dt 0.002 --method RUNGE_KUTTA
```
などとプロジェクトファイルを再作成することになりそうです．

(ちなみにROSwikiの編集は少し前からスパム対策で自分のROSwikiアカウント名を管理人にコメントで教える形式になったようです)

closed 2016-04-08 by ishiguroJSK

snozawa, 2016-04-08:
階段のサンプルは古いサンプルになってて、
確かSTがシミュレーションモード＋torqueモードでやるより前のものだったので、
STを入れてなくて、STなしで実行していたきがします。

ishiguroJSK, 2016-04-08:
samplerobot-terrain-walk.lとかの中見たらST入ってなかったですね．
確かにsamplerobot-stabilizer.lの時は上に飛び跳ねてしまっていたのですが，
階段上りができていたのでてっきりST動いているんだと思ってました．
samplerobot-terrain-walk.lとかに素人知識でST入れて試してみましたけど，やはり伸び上がるように失敗してしまいました．
とりあえずJAXONのChoreonoid環境がSTありで動くようになったので，以降はそちらで遊んでみます．