start-jsk/rtmros_common #927: debのhrpsys_ros_bridgeのsamplerobot-interface.lを実行するとエラーが出る

ban-masa, 2016-04-18:
debのhrpsys_ros_bridgeのsamplerobot-interface.lを実行すると以下のエラーが出ました。必要なmsgが生成されていないようです。srcを落とすのではなくdebの方で解決することは出来るのでしょうか。
```
$ roseus /opt/ros/indigo/share/hrpsys_ros_bridge/euslisp/samplerobot-interface.l                                                                                                                                                        
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x4f22de0[16374] --> 0x5383c00[32748] top=3c7c
irtgl irtviewer 
EusLisp 9.18( 1.0.12) for Linux64 created on ip-172-31-4-234(Sun Apr 3 22:43:01 PDT 2016)
roseus ;; loading roseus("") on euslisp((9.18 ip-172-31-4-234 Sun Apr 3 22:43:01 PDT 2016  1.0.12))
eustf roseus_c_util [ERROR] [1460979058.525437602]: Could not find roseus messages for hrpsys_ros_bridge under (/opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh hrpsys_ros_bridge
;; can not find nil/msg/ directory for [hrpsys_ros_bridge] packages
[ INFO] [1460979058.525541824]: cell* ROSEUS_EXIT(context*, int, cell**)
```

k-okada, 2016-04-18:
どこかいいっこ hrpsys_ros_bridgeに依存するパッケージをもってきてcaktin make して上げる必要があります．
あるいは，
rosrun roseus generate-all-msg-srv.sh hrpsys_ros_bridge
とか
rosrun roseus generate-all-msg-srv.sh --all として食事に行く
とか，
jade をつかう
と解決できると思います

ban-masa, 2016-04-18:
jadeの方のパッケージならばdebでのinstall後にこのerrorが出ないのですね。indigoの方でも対処してほしいと思うのですが、indigoのパッケージの保守は終了しているのでしょうか。

k-okada, 2016-04-18:
jadeからroseusが"公式"のrosクライアントライブラリになりました．

```
$ dpkg -L ros-jade-std-msgs  | grep String.l$
/opt/ros/jade/share/roseus/ros/std_msgs/msg/String.l
$ dpkg -L ros-indigo-std-msgs  | grep String.l$
```

closed 2016-11-15 by ban-masa