start-jsk/rtmros_common #928: hrpsys_ros_bridge/models/samplerobot.lが存在しない

ban-masa, 2016-04-18:
debのhrpsys_ros_bridgeの方にmodels/samplerobot.lが存在しないため、以下のようなエラーが出ました。ソースを落とさずに解決する方法はないでしょうか。
```
;; extending gcstack 0x4fe2c00[32738] --> 0x5a84cb0[65476] top=7dd2
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  0x54b0c70 in (apply #'ros::load-org-for-ros ros::fullname args)E: 
1.B1-irteusgl$ 
nil
1.B1-irteusgl$ ,btrace
*unbound*
2.B1-irteusgl$ ,fullname
*unbound*
3.B1-irteusgl$ ,ros::fullname
"/opt/ros/indigo/share/hrpsys_ros_bridge/models/samplerobot.l"
4.B1-irteusgl$ ls /opt/ros/indigo/share/hrpsys_ros_bridge/models/samplerobot.l
ls: cannot access /opt/ros/indigo/share/hrpsys_ros_bridge/models/samplerobot.l: No such file or directory
512
5.B1-irteusgl$ 5.irteusgl$ [ INFO] [1460980511.364332975]: cell* ROSEUS_EXIT(context*, int, cell**)
[/opt/ros/indigo/share/hrpsys_ros_bridge/euslisp]
$ ls /opt/ros/indigo/share/hrpsys_ros_bridge/models/samplerobot.l
ls: cannot access /opt/ros/indigo/share/hrpsys_ros_bridge/models/samplerobot.l: No such file or directory
[/opt/ros/indigo/share/hrpsys_ros_bridge/euslisp]
$ ls /opt/ros/indigo/share/hrpsys_ros_bridge/models/             
SampleRobot.RobotHardware.conf  SampleRobot.xml                       SampleRobot_nosim.conf   pa10.dae                     pa10_meshes                    sample4legrobot.yaml
SampleRobot.conf                SampleRobot_controller_config.yaml    SampleRobot_nosim.xml    pa10.urdf                    pa10_nosim.RobotHardware.conf  samplerobot.yaml
SampleRobot.dae                 SampleRobot_meshes                    pa10.RobotHardware.conf  pa10.xml                     pa10_nosim.conf                samplespecialjointrobot.yaml
SampleRobot.urdf                SampleRobot_nosim.RobotHardware.conf  pa10.conf                pa10_controller_config.yaml  pa10_nosim.xml
```

k-okada, 2016-04-18:
http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__hrpsys_ros_bridge__ubuntu_trusty_amd64__binary/16/consoleFull
```
05:13:40 -- [compile_robot_model.cmake]
compile_openhrp_model(/opt/ros/indigo/share/OpenHRP-3.1/idl/../sample/model/sample1.wrl
/opt/ros/indigo/share/OpenHRP-3.1/idl/../sample/model/sample1.wrl;SampleRobot;--conf-dt-option;0.002;--simulation-timestep-option;0.002;--conf-file-option;abc_leg_offset:
0,0.09,0;--conf-file-option;end_effectors:
lleg,LLEG_ANKLE_R,WAIST,0.0,0.0,-0.07,0.0,0.0,0.0,0.0,
rleg,RLEG_ANKLE_R,WAIST,0.0,0.0,-0.07,0.0,0.0,0.0,0.0,
larm,LARM_WRIST_P,CHEST,0.0,0,-0.12,0,1.0,0.0,1.5708,
rarm,RARM_WRIST_P,CHEST,0.0,0,-0.12,0,1.0,0.0,1.5708,;--conf-file-option;collision_pair:
RARM_WRIST_P:WAIST LARM_WRIST_P:WAIST RARM_WRIST_P:RLEG_HIP_R
LARM_WRIST_P:LLEG_HIP_R RARM_WRIST_R:RLEG_HIP_R
LARM_WRIST_R:LLEG_HIP_R;--conf-file-option;# SequencePlayer optional data
(contactStates x 4 + controlSwingTime x 4 (4 is lfsensor, rfsensor,
lhsensor, rhsensor);--conf-file-option;seq_optional_data_dim:
8;--conf-file-option;pdgains_sim_file_name:
/opt/ros/indigo/share/hrpsys/samples/SampleRobot/SampleRobot.PDgain.dat)
05:13:40 --  wrl file
/opt/ros/indigo/share/OpenHRP-3.1/idl/../sample/model/sample1.wrl
05:13:40 -- yaml file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/samplerobot.yaml
05:13:40 --  dae file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/SampleRobot.dae
05:13:40 -- urdf file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/SampleRobot.urdf
05:13:40 -- lisp file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/samplerobot.l
05:13:40 --  xml file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/SampleRobot.xml
05:13:40 --
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/SampleRobot_nosim.xml
05:13:40 -- config file
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/models/SampleRobot_controller_config.yaml
05:13:40 -- Output
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/build/pa10-interface.l
05:13:40 -- Output
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1/build/samplerobot-interface.l
05:13:40 -- Configuring done
05:13:40 CMake Warning at cmake/compile_robot_model.cmake:301 (message):
05:13:40   could not find /lib/euscollada/collada2eus
05:13:40 Call Stack (most recent call first):
05:13:40   cmake/compile_robot_model.cmake:91 (get_collada2eus)
05:13:40   catkin.cmake:167 (compile_openhrp_model)
05:13:40   CMakeLists.txt:26 (include)
05:13:40
05:13:40
05:13:40 Generate launch files for pa10
05:13:40   PROJECT_PKG_NAME = hrpsys_ros_bridge
05:13:40         MODEL_FILE = $(find
openhrp3)/share/OpenHRP-3.1/sample/model/PA10/pa10.main.wrl
05:13:40              ROBOT = pa10
05:13:40 hrpsys_ros_bridge_PACKAGE_PATH =
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1
05:13:40 Generate launch files for SampleRobot
05:13:40   PROJECT_PKG_NAME = hrpsys_ros_bridge
05:13:40         MODEL_FILE = $(find
openhrp3)/share/OpenHRP-3.1/sample/model/sample1.wrl
05:13:40              ROBOT = SampleRobot
05:13:40 hrpsys_ros_bridge_PACKAGE_PATH =
/tmp/binarydeb/ros-indigo-hrpsys-ros-bridge-1.3.1
05:13:42 -- Generating done
```

ということで，deb 作るときに失敗していますね．

k-okada, 2016-04-18:
ああ，ちがう，euscollada に依存するとhrpsysがeusに依存するということになって，さすがにやり過ぎだろう．と思ったんだ．`
rosrun euscollada collada2eus `rospack find
hrpsys_ros_bridge`/models/SampleRobot.dae `rospackn find
hrpsys_ros_bridge`/samplerobot.yaml `rospack find
hrpsys_ros_bridge`/models/samplerobot.l`
 でつくるしかないか

ban-masa, 2016-04-18:
事情があって意図的にsamplerobot.lが入っていないのですね。debのhrpsys_ros_bridgeをeuslispで動かしたい場合は手動でsamplerobot.lを生成せざるをえないという問題は解決できないということですか。ROSWikiのtutorial http://wiki.ros.org/rtmros_common/Tutorials/WorkingWithEusLisp にはそのような記載はなかったのですが、追記か何かしておくべきでしょうか。

k-okada, 2016-04-18:
手動でやったらうまくいったかな？それで上手く行くなら書いておいてくれると嬉しいです．

k-okada, 2016-04-26:
1.3.2 ではsamplerobot.l が入るようにしたつもりです．明日にはdebが配られると思うのでお楽しみに．

snozawa, 2016-04-26:
> 1.3.2 ではsamplerobot.l が入るようにしたつもりです．明日にはdebが配られると思うのでお楽しみに．

これはeuscolladaか何かのdependが入ったんでしょうか？
それともsamplerobot.lの生成物がコミットされてるんでしょうか？

k-okada, 2016-04-26:
build dependだけにいれました．
ソース落として rosdepinstall すると入るeuscolladaがはいる．
apt-get ros-indigo-hrpsys-ros-bridge は入らない．

snozawa, 2016-08-16:
こちら多分OKそうです。

closed 2016-08-16 by snozawa