start-jsk/rtmros_common #936: rtmlaunch.pyでimportエラーが発生する

iory, 2016-04-21:
rtmros_commonをsourceから入れてcatkin buildすると，
rtmlaunchで下記のようなエラーが発生します．

```
iory@crab : ~/catkin_ws/hrp2/src/rtmros_common/hrpsys_ros_bridge $ rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
Traceback (most recent call last):
  File "/home/iory/catkin_ws/hrp2/devel/bin/rtmlaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 28, in <module>
  File "/home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/src/openrtm_tools/rtmlaunch.py", line 15, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver'
```

iory, 2016-04-21:
ここまでのcommitではrtmlaunchが起動できています．
719f4f79bb2f749aab92aaa810e9bf51b73cf03f

比較
https://github.com/start-jsk/rtmros_common/compare/719f4f79bb2f749aab92aaa810e9bf51b73cf03f...master

iory, 2016-04-21:
commitというよりmanifest.xmlがあるかないかで変わるようです．
https://github.com/start-jsk/rtmros_common/compare/719f4f79bb2f749aab92aaa810e9bf51b73cf03f...master#diff-8f9c3a0467bfae1e16636bad7fa23fc7

snozawa, 2016-04-21:
> commitというよりmanifest.xmlがあるかないかで変わるようです．

これは、
- manifest.xmlの有無

で、
- うまくきどうできるorできない

がどうなりますか？
こちらの想定としては、実質不要なファイルなのであっても動いてくれるはず

iory, 2016-04-21:
manifest.xmlがあるとrtmlaunchが正常に起動しますが，
manifest.xmlがないと以下のエラーで起動できないです．
```
iory@crab : ~/catkin_ws/hrp2/src/rtmros_common/hrpsys_ros_bridge $ rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
Traceback (most recent call last):
  File "/home/iory/catkin_ws/hrp2/devel/bin/rtmlaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 28, in <module>
  File "/home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/src/openrtm_tools/rtmlaunch.py", line 15, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver'
```
pathかなにかが良くないでしょうか．

iory, 2016-04-21:
rtmros_common/openrtm_tools以下にmanifest.xmlがある場合，起動できる．
ない場合，起動できないです．

k-okada, 2016-04-21:
rtmros_common/openrtm_tools/package.xml はあるよね．
pythonpathを作るときにこれを見に行くから重要情報であはる．
ls -al ~/catkni_ws/hrp2/src
と
rospack list | grep catkin_ws
をしりたいです

k-okada, 2016-04-21:
rtmlaunch openrtm_tools test-rtmlaunch.test
は動く？

iory, 2016-04-21:
```
iory@crab : ~/catkin_ws/hrp2/src $ ls      
hrpsys  iai_kinect2  iory_demo  openhrp3  rtm-ros-robotics  rtmros_common
```
rospack listの中身が以下で，
```
iory@crab : ~/catkin_ws/hrp2 $ rospack list | grep catkin_ws                     
hironx_tutorial /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hironx_tutorial
hrp2_models /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models
hrpsys /home/iory/catkin_ws/hrp2/src/hrpsys
hrpsys_gazebo_tutorials /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials
hrpsys_ros_bridge /home/iory/catkin_ws/hrp2/src/rtmros_common/hrpsys_ros_bridge
hrpsys_ros_bridge_tutorials /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials
hrpsys_tools /home/iory/catkin_ws/hrp2/src/rtmros_common/hrpsys_tools
hrpsys_tutorials /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_tutorials
iory_demo /home/iory/catkin_ws/hrp2/src/iory_demo
jsk_hrp2_ros_bridge /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/jsk_hrp2_ros_bridge
jsk_models /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/jsk_models
kinect2_bridge /home/iory/catkin_ws/hrp2/src/iai_kinect2/kinect2_bridge
kinect2_calibration /home/iory/catkin_ws/hrp2/src/iai_kinect2/kinect2_calibration
kinect2_registration /home/iory/catkin_ws/hrp2/src/iai_kinect2/kinect2_registration
kinect2_viewer /home/iory/catkin_ws/hrp2/src/iai_kinect2/kinect2_viewer
openhrp3 /home/iory/catkin_ws/hrp2/src/openhrp3
openhrp3_tutorials /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/openhrp3_tutorials
openrtm_ros_bridge /home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_ros_bridge
openrtm_tools /home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_tools
rosnode_rtc /home/iory/catkin_ws/hrp2/src/rtmros_common/rosnode_rtc
rtmbuild /home/iory/catkin_ws/hrp2/src/rtmros_common/rtmbuild
```

iory, 2016-04-21:
動かないです．
```
iory@crab : ~/catkin_ws/hrp2/src $ rtmlaunch openrtm_tools test-rtmlaunch.test 
Traceback (most recent call last):
  File "/home/iory/catkin_ws/hrp2/devel/bin/rtmlaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 28, in <module>
  File "/home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/src/openrtm_tools/rtmlaunch.py", line 15, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver'
```

k-okada, 2016-04-21:
```
k-okada@kokada-t440s:~/catkin_ws/ws_rtmros/src/rtmros_common/openrtm_tools$ which rtmlaunch openrtm_tools test-rtmlaunch.test
/home/k-okada/catkin_ws/ws_rtmros/devel/bin/rtmlaunch
k-okada@kokada-t440s:~/catkin_ws/ws_rtmros/src/rtmros_common/openrtm_tools$ rospack list | grep catkin_ws
hrpsys_ros_bridge /home/k-okada/catkin_ws/ws_rtmros/src/rtmros_common/hrpsys_ros_bridge
hrpsys_tools /home/k-okada/catkin_ws/ws_rtmros/src/rtmros_common/hrpsys_tools
openrtm_tools /home/k-okada/catkin_ws/ws_rtmros/src/rtmros_common/openrtm_tools
rtmbuild /home/k-okada/catkin_ws/ws_rtmros/src/rtmros_common/rtmbuild
k-okada@kokada-t440s:~/catkin_ws/ws_rtmros/src/rtmros_common/openrtm_tools$ rtmlaunch openrtm_tools test-rtmlaunch.test
[rtmlaunch] Start omniNames at port 15005 

Thu Apr 21 20:58:29 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003139322e3136382e3131312e323800009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c00000001000000010001000100000001000105090101000100000009010100035454410800000065c0185701005b0b
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
('log_dir = ', '/home/k-okada/.ros/log/2326df66-07b8-11e6-b087-28b2bde39261')
('latest_dir = ', '/home/k-okada/.ros/log/latest')
... logging to /home/k-okada/.ros/log/2326df66-07b8-11e6-b087-28b2bde39261/roslaunch-kokada-t440s-23285.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/k-okada/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://kokada-t440s:47272/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.19

NODES
  /
    consumer (openrtm_tools/MyServiceConsumerComp)
    provider (openrtm_tools/MyServiceProviderComp)
    rtmlaunch_py (openrtm_tools/rtmlaunch.py)
    seqin (openrtm_tools/SeqInComp)
    seqout (openrtm_tools/SeqOutComp)
    start_omninames (rtmbuild/start_omninames.sh)

WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[start_omninames-1]: started with pid [23324]

Thu Apr 21 20:58:30 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003139322e3136382e3131312e32380000f90a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c00000001000000010001000100000001000105090101000100000009010100035454410800000066c0185701005b1e
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
process[provider-2]: started with pid [23327]
process[consumer-3]: started with pid [23339]
process[seqin-4]: started with pid [23362]
process[seqout-5]: started with pid [23401]
```
rtmros_commonだけのworkspaceをつくるとどうなるかな

iory, 2016-04-21:
ダメですね`
``
 iory@crab : ~/catkin_ws/rtmros_common : ls
build  devel  logs  src
 iory@crab : ~/catkin_ws/rtmros_common : ls src
rtmros_common
 iory@crab : ~/catkin_ws/rtmros_common :
 iory@crab : ~/catkin_ws/rtmros_common : rtmlaunch openrtm_tools test-rtmlaunch.test
Traceback (most recent call last):
  File "/home/iory/catkin_ws/rtmros_common/devel/bin/rtmlaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 28, in <module>
  File "/home/iory/catkin_ws/rtmros_common/src/rtmros_common/openrtm_tools/src/openrtm_tools/rtmlaunch.py", line 15, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver

iory, 2016-04-21:
```
 iory@crab : ~/catkin_ws/rtmros_common : source devel/setup.zsh
 iory@crab : ~/catkin_ws/rtmros_common : which rtmlaunch openrtm_tools test-rtmlaunch.test
/home/iory/catkin_ws/rtmros_common/devel/bin/rtmlaunch
openrtm_tools not found
test-rtmlaunch.test not found
iory@crab : ~/catkin_ws/rtmros_common : rospack list | grep catkin_ws
hrpsys_ros_bridge /home/iory/catkin_ws/rtmros_common/src/rtmros_common/hrpsys_ros_bridge
hrpsys_tools /home/iory/catkin_ws/rtmros_common/src/rtmros_common/hrpsys_tools
openrtm_ros_bridge /home/iory/catkin_ws/rtmros_common/src/rtmros_common/openrtm_ros_bridge
openrtm_tools /home/iory/catkin_ws/rtmros_common/src/rtmros_common/openrtm_tools
rosnode_rtc /home/iory/catkin_ws/rtmros_common/src/rtmros_common/rosnode_rtc
rtmbuild /home/iory/catkin_ws/rtmros_common/src/rtmros_common/rtmbuild
```

iory, 2016-04-21:
この状態でもmanifest.xmlを持ってくると,
rtmlaunch openrtm_tools test-rtmlaunch.testは起動します．

k-okada, 2016-04-21:
echo $ROS_PACKAGE_PATH は？

iory, 2016-04-21:
```
/home/iory/catkin_ws/rtmros_common/src/rtmros_common/hrpsys_tools:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/openrtm_tools:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/rosnode_rtc:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/rtmbuild:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/hrpsys_ros_bridge:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/openrtm_ros_bridge:/home/iory/catkin_ws/rtmros_common/src/rtmros_common/rtmros_common:/opt/ros/indigo/share:/opt/ros/indigo/stacks
```
です．

k-okada, 2016-04-21:
sudo updatedb
 locate ComponentObserver
 locate OpenRTM__POA

iory, 2016-04-21:
```
 iory@crab : ~ : locate ComponentObserver
/home/iory/catkin_ws/openrtm_aist/devel/lib/openrtm-1.1/sdo/ComponentObserverConsumer.a
/home/iory/catkin_ws/openrtm_aist/devel/lib/openrtm-1.1/sdo/ComponentObserverConsumer.la
/home/iory/catkin_ws/openrtm_aist/devel/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so
/home/iory/catkin_ws/openrtm_aist/devel/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so.0
/home/iory/catkin_ws/openrtm_aist/devel/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so.0.0.0
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserver.idl
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverConsumer.cpp
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverConsumer.h
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverConsumer.la
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverConsumer.lo
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverConsumer.o
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverSkel.cpp
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverSkel.h
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverStub.cpp
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverStub.h
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverStub.lo
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/ComponentObserverStub.o
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/libComponentObserverStub.la
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.deps/ComponentObserverConsumer.Plo
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.deps/ComponentObserverStub.Plo
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.a
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.la
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.lai
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.o
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.so
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.so.0
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.so.0.0.0
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverConsumer.so.0.0.0T
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/ComponentObserverStub.o
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/libComponentObserverStub.a
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/.libs/libComponentObserverStub.la
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/idl/ComponentObserver.hh
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/idl/ComponentObserver.idl
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/idl/ComponentObserverDynSK.cc
/home/iory/catkin_ws/openrtm_aist/src/openrtm_aist/src/ext/sdo/observer/idl/ComponentObserverSK.cc
/opt/ros/indigo/lib/openrtm-1.1/sdo/ComponentObserverConsumer.a
/opt/ros/indigo/lib/openrtm-1.1/sdo/ComponentObserverConsumer.la
/opt/ros/indigo/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so
/opt/ros/indigo/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so.0
/opt/ros/indigo/lib/openrtm-1.1/sdo/ComponentObserverConsumer.so.0.0.0
/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/ext/sdo/observer/ComponentObserverConsumer.py
/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/ext/sdo/observer/ComponentObserverConsumer.pyc
/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/ComponentObserver_idl.py
/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/ComponentObserver_idl.pyc
/usr/lib/python2.7/dist-packages/OpenRTM_aist/ext/sdo/observer/ComponentObserver.idl
/usr/lib/python2.7/dist-packages/OpenRTM_aist/ext/sdo/observer/ComponentObserverConsumer.py
/usr/lib/x86_64-linux-gnu/openrtm-1.1/sdo/ComponentObserverConsumer.a
/usr/lib/x86_64-linux-gnu/openrtm-1.1/sdo/ComponentObserverConsumer.la
/usr/lib/x86_64-linux-gnu/openrtm-1.1/sdo/ComponentObserverConsumer.so
/usr/lib/x86_64-linux-gnu/openrtm-1.1/sdo/ComponentObserverConsumer.so.0
/usr/lib/x86_64-linux-gnu/openrtm-1.1/sdo/ComponentObserverConsumer.so.0.0.0
 iory@crab : ~ : locate OpenRTM__POA     
/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM__POA
/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM__POA/__init__.py
/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM__POA/__init__.pyc
/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/OpenRTM__POA
/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/OpenRTM__POA/__init__.py
/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/OpenRTM__POA/__init__.pyc
/usr/lib/python2.7/dist-packages/OpenRTM__POA
/usr/lib/python2.7/dist-packages/OpenRTM__POA/__init__.py
/usr/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM__POA
/usr/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM__POA/__init__.py
```

k-okada, 2016-04-21:
dpkg -S /usr/lib/python2.7/dist-packages/OpenRTM__POA

iory, 2016-04-21:
iory@crab : ~ : dpkg -S /usr/lib/python2.7/dist-packages/OpenRTM__POA
openrtm-aist-python: /usr/lib/python2.7/dist-packages/OpenRTM__POA
 iory@crab : ~ : source ~/catkin_ws/rtmros_common/devel/setup.zsh
 iory@crab : ~ : dpkg -S /usr/lib/python2.7/dist-packages/OpenRTM__POA
openrtm-aist-python: /usr/lib/python2.7/dist-packages/OpenRTM__POA
```

k-okada, 2016-04-21:
sudo apt-get remove openrtm-aist-python

iory, 2016-04-21:
```
 iory@crab  ~  sudo apt-get remove openrtm-aist-python
[sudo] password for iory: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  openrtm-aist-python
0 upgraded, 0 newly installed, 1 to remove and 6 not upgraded.
After this operation, 2,523 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 654067 files and directories currently installed.)
Removing openrtm-aist-python (1.1.0+20160406+346+8~ubuntu14.04.1) ...

 iory@crab  ~  dpkg -S /usr/lib/python2.7/dist-packages/OpenRTM__POA
dpkg-query: no path found matching pattern /usr/lib/python2.7/dist-packages/OpenRTM__POA
```

k-okada, 2016-04-21:
これでrtmlaunchうごかないかな

iory, 2016-04-21:
動くようになりました．
debのpackageになにかしらしてしまっていたということでしょうか

k-okada, 2016-04-21:
rosではいるopenrtm-pythonはros-indigo-openrtm-python，一方JVRCかなにかのときにopenrtm-pythonが松阪さんのppaから入るようになっている．
で問題はこの２つのバージョンが微妙に違うので，動かないことになっているきがする．
aptitude show ros-indigo-openrtm-aist-python
aptitude show openrtm-aist-python
と，もう一度openrtm-aist-pythonをinstallして
diff -r /opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/ /usr/lib/python2.7/dist-packages/OpenRTM_aist/
みたいなので差分を調べたい．あと，どこかのMLで一回この話題出ている気がするので検索してみる．

iory, 2016-04-21:
なるほど．
たしかに昨日jvrc用のpackageをいれていました．
もう少し調べてみます．

iory, 2016-04-21:
memo
aptitude hsow ros-indigo-openrtm-aist-python
aptitude show opnertm-aist-pythonの結果
```
 iory@crab  ~  aptitude show ros-indigo-openrtm-aist-python
Package: ros-indigo-openrtm-aist-python  
State: installed
Automatically installed: no
Version: 1.1.0-6trusty-20160109-141419-0800
Priority: extra
Section: misc
Maintainer: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Architecture: amd64
Uncompressed Size: 3,356 k
Depends: omniidl-python, python-omniorb, python-omniorb-omg
Conflicts: ros-indigo-openrtm-aist-python
Description: Python binding of OpenRTM-AIST (see openrtm_aist for further information). OpenRTM-aist is an RT-Middleware-baseed, component-oriented software platform to robotics development that is made and maintained in AIST (National Institute of Advanced Industrial Science and Technology) in Japan (excerpts from here)
 
Homepage: http://ros.org/wiki/openrtm_aist

 iory@crab  ~  aptitude show openrtm-aist-python
Package: openrtm-aist-python             
New: yes
State: not installed
Version: 1.1.0+20160406+346+8~ubuntu14.04.1
Priority: extra
Section: science
Maintainer: Noriaki Ando <n-ando@aist.go.jp>
Architecture: amd64
Uncompressed Size: 2,523 k
Depends: omniidl-python, python-omniorb
Conflicts: openrtm-aist-python
Description: OpenRTM-aist is a reference implementation of RT-Middleware
 and RT-Component framework. RT-Component (RTC) is a component model
 standardized in OMG (Object Management Group) as Robotic Technology Component
 Specification 1.0 (formal/08-04-04, http://www.omg.org/spec/RTC/1.0).
 OpenRTM-aist is being developed and distributed by Intelligent Systems Research
 Institute, National Institute of Advanced Industrial Science and Technology
 (AIST), Japan. Please see http://openrtm.org/ for more details.
```

iory, 2016-04-21:
MLは以下のようです．
https://groups.google.com/forum/#!msg/rtm-ros-robotics/hXwmjDFxiVY/rohU62RlMkUJ

YoheiKakiuchi, 2016-04-21:
これですか？ タイトルとはちょっと違うけど中身は、OpenRTMのdebの話になっていっていたような。
https://github.com/fkanehiro/hrpsys-base/issues/764

iory, 2016-04-21:
ありがとうございます．そのissueでも言及されていますね．
結局どうするのがいいでしょうか．
choreonoidを入れるときにopenrtm-aist-pythonを入れてしまうと，
hrpsysを立ち上げる時に問題となるかもしれないのですが

k-okada, 2016-04-21:
choreonoid使わない、が現実的な気がしますが、それ置いておいて
manifestを入れると動くのは不思議
落ちる直前あたりでprint(sys.path)辺りをすると何が違うだろうか

iory, 2016-04-21:
importのところで，そもそもOpenRTM__POAがComponentObserverを持っていないで落ちていて，
普通にprint debugをすると，以下のような変更で(rtmlaunch)
```
 cosnames = "omniNames"
 port_number = 15005
-
+print("start===========================================================================================================")
+import sys
+print(sys.path)
 from openrtm_tools import rtmstart, rtmlaunch
+print("end=============================================================================================================")
 import roslaunch
 ```

```
 iory@crab  ~/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/scripts   master ●  rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
start===========================================================================================================
['/home/iory/catkin_ws/hrp2/devel/bin', '/usr/local/lib/python2.7/dist-packages/greenlet-0.4.9-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/gevent-1.1.0-py2.7-linux-x86_64.egg', '/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages', '/opt/ros/indigo/lib/python2.7/dist-packages', '/usr/local/Cellar/opencv3/3.0.0/lib/python3.4/site-packages', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/OpenRTM_aist', '/usr/lib/python2.7/dist-packages/OpenRTM_aist/utils', '/usr/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/omniORB/COS', '/usr/lib/python2.7/dist-packages/gst-0.10', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/pymodules/python2.7', '/usr/lib/python2.7/dist-packages/ubuntu-sso-client', '/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode']
Traceback (most recent call last):
  File "/home/iory/catkin_ws/hrp2/devel/bin/rtmlaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 30, in <module>
  File "/home/iory/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/src/openrtm_tools/rtmlaunch.py", line 15, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver'
```

iory, 2016-04-21:
いかが正常に起動した場合です．
```
 iory@crab  ~/catkin_ws/hrp2/src/rtmros_common/openrtm_tools/scripts   master ●  rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
start===========================================================================================================
['/home/iory/catkin_ws/hrp2/devel/bin', '/usr/local/lib/python2.7/dist-packages/greenlet-0.4.9-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/gevent-1.1.0-py2.7-linux-x86_64.egg', '/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages', '/opt/ros/indigo/lib/python2.7/dist-packages', '/usr/local/Cellar/opencv3/3.0.0/lib/python3.4/site-packages', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/omniORB/COS', '/usr/lib/python2.7/dist-packages/gst-0.10', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/pymodules/python2.7', '/usr/lib/python2.7/dist-packages/ubuntu-sso-client', '/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode']
end=============================================================================================================
[rtmlaunch] Start omniNames at port 15005 

Fri Apr 22 02:30:14 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000c0000003135372e38322e352e3330009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000260e1957010071b4
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
```

iory, 2016-04-21:
diffをとると，
```
 iory@crab  /tmp  diff without-openrtm-aist-python.txt with-openrtm-aist-python.txt 
13a14,16
> /usr/lib/python2.7/dist-packages/OpenRTM_aist
> /usr/lib/python2.7/dist-packages/OpenRTM_aist/utils
> /usr/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL
```
です．

k-okada, 2016-04-22:
rtmlaunch.py で
```
+ from rtctree.rtmidl import OpenRTM__POA
  import rtctree
```
とすると上手く動く？

iory, 2016-04-22:
それで動くようになりました

k-okada, 2016-04-22:
https://github.com/start-jsk/rtmros_common/commit/94975a81eadbf251a85b2fd3c77137b1f96f248b
をrevertするのが正解？

iory, 2016-04-22:
そっちのほうがいいですかね．
revertしてPR送ります

iory, 2016-04-22:
@k-okada 
travisが失敗しているようですが，
最後に
```
The job exceeded the maximum time limit for jobs, and has been terminated.
```
と出ているのですが，これはどうすればいいでしょうか？

k-okada, 2016-04-22:
ときどきあるので，travisをリスタートして下さい．
https://github.com/jsk-ros-pkg/jsk_common

iory, 2016-04-22:
はい，travis通りました．
mergeされ次第closeします．

PR #938 (2016-04-22): Revert "workaround for ros-groovy-rqt-top installs wrong(?) psutil"
This reverts commit 94975a81eadbf251a85b2fd3c77137b1f96f248b. To fix #936
closed 2016-04-22 by k-okada