start-jsk/rtmros_common #975: roseus test in hrpsys_ros_bridge is not called in jenkins

mmurooka, 2016-10-17:
[hrpsys_ros_bridge内のroseusのテスト](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L269-L270) がjenkinsのテストで呼ばれていないことが，https://github.com/start-jsk/rtmros_common/pull/972#issuecomment-254111547 で分かりました．
travisからjenkinsに環境変数`IS_EUSLISP_TRAVIS_TEST`を渡していないことが原因です（https://github.com/start-jsk/rtmros_common/pull/973 ）．

関連するPRが複数あるので解決策と合わせて整理すると（上から順に望ましい？），

（１） [環境変数のif文](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L268)をそもそも無くす．
　このif文は何を目的としているでしょうか．
（２） cmake-argsをtravisからjenkinsに渡すようにして，環境変数`IS_EUSLISP_TRAVIS_TEST`の代わりにこれを使う．（https://github.com/start-jsk/rtmros_common/pull/973#issuecomment-254132210 ）
　（４）案よりも汎用的
（３） すべてのテストをtravisで走らせる． https://github.com/start-jsk/rtmros_common/pull/974
　hydroやkineticではtravisが使えない？
（４） jsk_travis/travis_jenkins.pyで環境変数`IS_EUSLISP_TRAVIS_TEST`をjenkinsに渡す．（https://github.com/jsk-ros-pkg/jsk_travis/blob/master/travis_jenkins.py#L25-L40 ）
　他のリポジトリでも使うところに個別のことを書かないほうが良い．

また，そもそもroseusのテストが走るようにすると49分くらい経過して
> The log length has exceeded the limit of 4 MB (this usually means that the test suite is raising the same exception over and over).

と出て終わってしまったので，テストが走るようになったとして，travis/jenkins上でできるように何か変更が必要．
https://travis-ci.org/start-jsk/rtmros_common/jobs/168202321

k-okada, 2016-10-18:
2016年10月18日 2:20 Masaki Murooka <notifications@github.com>:

> （１） 環境変数のif文
> <https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L268>
> をそもそも無くす．
> このif文は何を目的としているでしょうか．
>
途中でcaktin.cmake を CMakeLists.txt にリネームしているのでたどりづらいですが
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/181
です．なくしていいのかな．

--
◉ Kei Okada

mmurooka, 2016-10-18:
> jsk-ros-pkg/jsk_model_tools#181
です．

は別のリンクとの間違いでしょうか．違ったらすみません．

snozawa, 2016-10-18:
>（１） 環境変数のif文をそもそも無くす． #976
>　このif文は何を目的としているでしょうか．

roseusのif文がはいったら、元々eus docが呼ばれるようになってました。
その時にroseus+hrpsysのテストが走らないようにするために入っています。
roseus+hrpsysのテストのときは、hrpsys などをソースでいれてorigin/masterでテストしてたとおもいます。
eus docを作るときにはそうでなくて、hrpsysなどがaptではいるようにしています。

snozawa, 2016-10-18:
最近の情勢がわかってないので初歩的な質問なのですが、
1. 今はtravisが走ったときに、travis上とjenkins上の両方でテストが走る？
2. jenkinsを呼ぶときには、元々travis上で設定されてた環境変数などは渡されない？

travisとjenkinsが透過的に使えることを狙うとすると、travis上で使えてる環境変数は
jenkinsでもセットすると良いような気がしました。

mmurooka, 2016-10-18:
> 1. 今はtravisが走ったときに、travis上とjenkins上の両方でテストが走る？

https://github.com/start-jsk/rtmros_common/blob/master/.travis.yml#L20-L37
でUSE_TRAVISとUSE_JENKINSのどちらかがtrueになっていて切り分けられています．

> 2. jenkinsを呼ぶときには、元々travis上で設定されてた環境変数などは渡されない？

はい．
https://github.com/jsk-ros-pkg/jsk_travis/blob/master/travis_jenkins.py#L25-L40
で個別にtravisからjenkinsに渡す環境変数をひとつずつ列挙しています．
すべての環境変数を渡せるといいですが，そのやり方はわからないです．

snozawa, 2016-10-18:
なるほど、環境変数はわたらないんですね。

mmurooka, 2016-10-18:
> （２） cmake-argsをtravisからjenkinsに渡すようにして，環境変数IS_EUSLISP_TRAVIS_TESTの代わりにこれを使う．（#973 (comment) ）

で，環境変数を全部渡すということができなくても，cmake-argsに相当する変数のみを渡せれば，いいというのが案２です．

snozawa, 2016-10-18:
cmake-argsはどの工程で呼ばれてますか？

k-okada, 2016-10-18:
2016-10-18 12:11 GMT+09:00 Masaki Murooka <notifications@github.com>:

> jsk-ros-pkg/jsk_model_tools#181
> <https://github.com/jsk-ros-pkg/jsk_model_tools/pull/181>
> です．
>
> は別のリンクとの間違いでしょうか．違ったらすみません．
>
ああ、そうです
https://github.com/start-jsk/rtmros_common/pull/806
かな．（mv するとリンクたどりづらくて．．．．）

--
◉ Kei Okada

snozawa, 2016-10-19:
https://github.com/start-jsk/rtmros_common/issues/975#issuecomment-254391996
のそもそものこのif文をなくすとしたら、おそらく
eusdocのtravis上のジョブがfailせず、
hrpsysがorigin/masterでhrpsys+rosbridge+euslispがはしる
というようになるようなif文であれば、環境変数がなくてもできると思います。

@mmurooka 
一回単純に環境変数のif文をなくして、travis上のテストをはしらせて、
どこがどういうふうにfailするかをみてみると良いかもしれません。

mmurooka, 2016-10-19:
> 一回単純に環境変数のif文をなくして、travis上のテストをはしらせて、
どこがどういうふうにfailするかをみてみると良いかもしれません。

はい，こちらになります．https://github.com/start-jsk/rtmros_common/pull/976