start-jsk/rtmros_common #977: Authority for setting label

mmurooka, 2016-10-18:
@k-okada , 
In jsk-ros-pkg/jsk_common, we can set label for PR even though we cannot merge PR (f.g. https://github.com/jsk-ros-pkg/jsk_recognition/pull/1904 ).
Is it possible to give us such authority for this repository too?

I'd like to set label to the following PRs: https://github.com/start-jsk/rtmros_common/pull/976, https://github.com/start-jsk/rtmros_common/pull/974, https://github.com/start-jsk/rtmros_common/pull/973 .

k-okada, 2016-10-18:
> merge PR (f.g. jsk-ros-pkg/jsk_recognition#1904

changed setting, please check this

mmurooka, 2016-10-18:
Thank you. It worked.

snozawa, 2016-10-25:
Authorityの設定はどこみたらよかったでしょうか？
私のauthority権限がかわってるかもしれません（mergeボタンのところなどにnot authorizedとでてきます）

snozawa, 2016-10-25:
今みたらauthorizedされてました。

k-okada, 2016-10-25:
直ったかな

snozawa, 2016-10-25:
作業ありがとうございます。

昨日午後くらいに一瞬なおったのですが、
https://github.com/start-jsk/rtmros_common/pull/981
https://github.com/start-jsk/rtmros_common/pull/983
をみると、mergeボタンのところに
```
- Review required
At least one approved review is required by reviewers with write access.
- All checks have passed
1 successful check
Details
Required
continuous-integration/travis-ci/pr — The Travis CI build passed
- Merging is blocked
Merging can be performed automatically with one approved review. 
```
とでてました。
mergeがblockされてるとでているようでした。
こちらも何か変更がありましたでしょうか。

mmurooka, 2016-10-27:
@k-okada 
ラベルをつける権限とレビューを必須にすることは独立に設定できて，
今回はそれが同時に変更されたという理解であっていますか．

snozawa, 2016-10-27:
@k-okada さん
運用方針に変更が加えるときには、変更内容一式をアナウンスしていただけると助かります。
権限変更などgithub上での設定をかえるときにもissueをたてる、というのはいかがでしょうか。