start-jsk/rtmros_common #985: :angle-vector-sequence-fullの引数仕様

snozawa, 2016-10-31:
@terasawaさん
https://github.com/start-jsk/rtmros_common/pull/954
eusの:angle-vector-sequence-fullの引数仕様ですが僕がmergeしてしまったのですが，
大事な議論を忘れてました．

:angle-vector系のメソッドは関節角が[deg], 時間が[ms]になってるので，このメソッドも
それに統一するのが良いように思いました．
他にも，ＺＭＰ，posなども[mm]が良いように思いました．

他にもeusの:angle-vector-sequence-fullを使ってる人はいますか？

terasawa, 2016-10-31:
もともとsetJointAnglesSequenceFullはSequencePlayerの補間が入るload-patternの上位互換のように思っていたので，
「:angle-vector」系というよりは「:load-pattern」系？のサービスを呼ぶイメージでこのメソッドを作りました．

実装する上ではSequenceファイルを作るときと同じように扱える（時間間隔を変えるだけ）という点で引数もSequenceファイルの単位系に合わせましたが，
このサービスを呼ぶメソッドを「:angle-vector」系と捉えて今後統一して使うなら確かに[deg]や[mm]にした方がいい気もします．

「load-pattern」系と捉えるにしても「:angle-vector-sequence-full」という命名は変更したほうがいいかもしれません．

私は現状はどちらに変わってもそんなには困らないので，
他に使っている方がいらっしゃらなければ，今後使う人が誤解が少なくて使いやすい方にしたいと思います．
（誤解がなくて本来のサービスの役割的に正解に近そうという点では「angle-vector」系メソッドの単位に合わせるのがいいかもしれませんね．）

snozawa, 2016-11-01:
> 「:angle-vector」系というよりは「:load-pattern」系？のサービスを呼ぶイメージでこのメソッドを作りました．

なるほど。
- setJointAngleXX系関数は、引数に指令値を与える。eusでは:angle-vector-xx系に対応
- loadPattern系は、指令値はファイル書き出しし、引数にはファイル名を与える。eusではload-pattern系に対応

とするような切り分けになってます。
なので、setJointAnglesSequenceFullはどちらかというとsetJointAngle系関数に含まれます。
またそのため、:angle-vector系メソッドを踏襲しておくのが無難そうですね。

時間あるときで良いので、:angle-vector-sequence-full関数の引数の仕様をかえるPRを作成してもらえると助かります。

terasawa, 2016-11-01:
了解しました。

実機での実験が落ち着いたら変更を加えてPRを出します。
angle-vectorと時間の単位はわかるのですが、ほかについてはangle-vector系のメソッドがどのような引数を取るべきかわからないのですが、どれか倣うべき例がありますか。
なければ、以下の左側で良いでしょうか？（←の右側は議論の余地がありそうな単位系です）
jpos：[deg]
tm：[ms]
-------------以下key変数
vel：[mm/s]     ← [mm/ms]？
torque:[Nmm]     ← [Nm]？
pos:[mm]
rpy:[deg]     ← [rad]？
acc:[mm/s^2]     ← [mm/(ms)^2]？
zmp:[mm]
wrench:[N, Nmm]     ← [N, Nm]？
optional:[]

terasawa, 2016-11-01:
あともうひとつ補足があって、
SequencePlayerのidlのコメントを見るに、
setJointAnglesSequenceFullのpossとzmpsは[mm]で与えると書いてあるのですが、
私がシミュレーションや実機で試した限り[m]が正しそうでして（つまりコメントが間違っている）、
他のvelsやaccsやwrenchesの実際に与えるべき値の次元がidlのコメントに書いてある次元で正しいのか不安なのですが、
そもそもwrenchなどの使い方がわからなくてコメントに書いてある単位系が正しいか確かめられていません。

snozawa, 2016-11-01:
お、そうですね。
SequencePlayerのidlは誤ってますね。
```
    /**
     * @brief Interpolate all joint angles, using duration specified by \em tm. Returns without waiting for whole sequence to be sent to robot. If this function called during the robot moves, it overwrite current goal.
     * @param jvss sequence of sequence of joint angles [rad]
     * @param vels sequence of sequence of velocities [rad/s]
     * @param accs sequence of sequence of torques [Nm]
     * @param poss sequence of weist pos [m]
     * @param rpys sequence of weist rpy [rad]
     * @param accs sequence of weist acc [m/ss]
     * @param zmps sequence of weist zmp [m]
     * @param wrenches sequence of wrenches [N, Nm]
     * @param optionss sequence of options []
     * @param tms sequence of duration [s]
     * @return true joint angles are set successfully, false otherwise
     */
    boolean setJointAnglesSequenceFull(in dSequenceSequence jvss, in dSequenceSequence vels,  in dSequenceSequence torques, in dSequenceSequence poss, in dSequenceSequence rpys, in dSequenceSequence accs, in dSequenceSequence zmps, in dSequenceSequence wrenchs, in dSequenceSequence optionals, in dSequence tms);
```
ですね。
直しときます。

eusのほうの引数は、
```
jpos：[deg]
tm：[ms]
-------------以下key変数
vel[deg/s]
torque:[Nm]
pos:[m]
rpy:[rad]
acc:[mm/s^2] ?
zmp:[m]
wrench:[N, Nm]
optional:[]
```
かな。

議論ポイントとしては、
- jposはeusが[deg]なので。
- tmは、:angle-vector系では[ms]であたえてるので
- velは、悩ましいけどhttps://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L310-L313 をみると[deg/s]のようで、また[deg]の角度を微分したら[deg/s]になるので、[deg/s]がよさげ
- torque, wrenchはeusでも[N], [Nm]なのでこれらで。
- pos, rpy : これは悩ましい。けどcoordsから直接得られるのが[mm], [rad]とかなので、[mm], [rad]がよさそう。ただし、ベースリンクを直接引数に与えた方が最も混乱なく使い勝手もよさそう
- zmp : これも[mm]かな。ただ、どこ相対のzmpかを変換必要
- optional : これらは無次元でいいです
- acc : どうしよう？？[mm/s^2]?

ベースリンクを引数にとれたら良いきがしたので、こちらでPR作ってみます。

snozawa, 2016-11-28:
@terasawaさん
https://github.com/start-jsk/rtmros_common/pull/988
でmergeされましたので，コードの変更とロボット実機の関連ＰＣのupdateをお願いします．
(単位系がかわってるので注意してupdateしてください)

closed 2016-11-28 by snozawa