start-jsk/rtmros_common #986: odomの原点

mttamtam, 2016-11-18:
@orikuma さんと確認したのですが、odomの原点が最初のlleg_end_coordsとrleg_end_coordsの中心にくるはずがずれていて、odom相対でロボットを見ると浮いています。
[image]
これは正しい挙動でしょうか？

YoheiKakiuchi, 2016-11-21:
これは、autobalancerを入れずに見た時の状態だよね？

https://github.com/jsk-ros-pkg/jsk_control/issues/624 この話に関わりますが、
ここでodomと言っているものはhrpsys-base/rtc/AutoBalancerのbaseTformを
そのまま出力しているように見えます。

ROSでの定義は以下にあって、
http://www.ros.org/reps/rep-0105.html
http://www.ros.org/reps/rep-0120.html

base_link, base_footprint, odom についての説明がありますが、
base_footprintは床面にあることが期待されているが、odomは明示的には
床面にあることは期待されていないように思います。

ですが、ここでの問題はbaseTformの初期値はvrmlのrootリンクの位置が出ていて
autobalancerを入れた瞬間に足のmidcoordsに飛んでしまうのでそれはそれでおかしい。
(odomは連続的であることと書いてある）

作戦としては、以下の通り
1-a) hrpsys-baseでodomを出すrtcをつくる
1-b) もしくは, AutoBalancerのbaseTformが常に正しいodomとして使えるように修正する
2) 1-aと同じものをrosのトピックから計算するノードを作る (or HrpsysSeqStateROSBridgeを修正)
3) 他に何かあるかな？

ちなみに、歩かない時のことも考えると、ここで議論している odomは
walk_odom (hrpsys_odom)のような名前であるべきではないかというのが
https://github.com/jsk-ros-pkg/jsk_control/issues/624 の議題です。

snozawa, 2016-11-21:
> ですが、ここでの問題はbaseTformの初期値はvrmlのrootリンクの位置が出ていて
> autobalancerを入れた瞬間に足のmidcoordsに飛んでしまうのでそれはそれでおかしい。
> (odomは連続的であることと書いてある）

足のmidcoordsに飛びますか？飛ばない仕様だと思います．
baseTformはrootの座標系がそのままでることになってます．
２脚ではこれはなおしていて，４脚はまだ治ってないかも．
start, stop, ２脚４脚にかかわらず，auto-balancerとしてはbaseTformは連続的な腰位置姿勢を出すのが正かいです．

odomは，文章が難しすぎて
http://www.ros.org/reps/rep-0105.html#odom
を読み解けてませんが，AutoBalancerからodomというポートを出すんだと思います．

YoheiKakiuchi, 2016-11-21:
> 足のmidcoordsに飛びますか？飛ばない仕様だと思います

誤解がありました。
autobalancerが入っていない時はdefaultのcoordsが出ている (姿勢を変えても動かない)
autobalancerを入れた瞬間に腰の位置に動く。

tfのodomはtformをodom->BODYとして出しているのでodomの位置は床(baseTformの原点)に出ます。
なので、autobalancerを入れるとodomは床 (midcoordsとは限らないかも）に飛びます。

snozawa, 2016-11-21:
@mttamtam 
ところで，この現象AutoBalancerを直しました，
という僕のhrpsys-baseのＰＲのコミット前に戻して再現するか確認できるかな．

snozawa, 2016-11-21:
あと，REPのodomの定義を読み込んで理解するのに当てられる時間が直近で厳しそうなので，
日本語か図で説明できる人がいると助かります．
そのうえで，飛ばないbaseTformがでていれば問題ないという理解であってますか？

yoshimalucky, 2016-11-21:
翻訳しました。

odomと呼ばれる座標系は世界座標系に固定された座標系です。odomの座標系に在る移動プラットホームの位置／姿勢は時間の経過とともに再現なくドリフトする可能性があります。このドリフトがodomを長期的にはグローバルな基準としては使い物にならなくしています。しかし、突発的な変化なく、odom座標系に在るロボットの位置姿勢の連続性は保障されており、odomに在る移動プラットホームは常に滑らかな軌道に沿って進みます。

一般的な設定においては、odom座標系は車輪のオドメトリ、ビジュアルオドメトリ、または、IMU等のオドメトリの入力によって計算されます。

オドメトリの座標系は正確で短期的なローカル座標系の基準としては使い物になりますが、長期的な基準としては、ドリフトがそれを脆弱な座標系という扱いにしてしまいます。

mttamtam, 2016-11-21:
@snozawa さん
戻してみたところ、画像のように足平に出てきました。
最初にstateholder?に原点の位置を入れないといけなくなったから、ということでしょうか。

[image]

YoheiKakiuchi, 2016-11-21:
> 誤解がありました。
> autobalancerが入っていない時はdefaultのcoordsが出ている (姿勢を変えても動かない)
> autobalancerを入れた瞬間に腰の位置に動く。
> tfのodomはtformをodom->BODYとして出しているのでodomの位置は床(baseTformの原点)に出ます。
> なので、autobalancerを入れるとodomは床 (midcoordsとは限らないかも）に飛びます。

全体の質問意図を理解出来てませんでした。
これは、以前(PR前)の仕様です。

PR後は、autobalancerを入れた時には, seqのbasePos, baseRPYを初期位置とするようになったので
autobalancerを入れるときに任意の位置になります。なのでodomも任意の位置。

> 飛ばないbaseTformがでていれば問題ないという理解であってますか？

それでいいはずです。初期位置は任意で良いようにROSの側を、必要なら、修正すべきでしょう。
ですが、まずは歩行中に飛ばないようにするのがいいように思います。
2足歩行は問題ないように思います。4足歩行は問題があるのかな？ > @mttamtam

autobalancerを切った時にどうなるかは、ROSの側で調整してもいいかもしれません。