start-jsk/rtmros_common #1016: 無くなったはずのsrvファイルがあって*ri*が作れなかった

mttamtam, 2017-06-26:
hrpsysまわりを最新にしてビルドしたあと、シミュレータを立ち上げ、
`(jaxon_red-init)`
とすると、
`/home/tamura/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable hrpsys_ros_bridge::openhrp_impedancecontrollerservice_getobjectturnarounddetectorparamrequest in (eval srv-request)`
と言われて、`*ri*`が作れませんでした。
当該の部分はOCTDとして別RTCになったはずなので、（https://github.com/fkanehiro/hrpsys-base/pull/1101）
どこか古い何かが残っているのだと思いcatkin cleanしてもう一度ビルドしなおしてもダメで、
`hrpsys_ros_bridge/srv`以下に
```
OpenHRP_ImpedanceControllerService_getObjectContactTurnaroundDetectorParam.srv
OpenHRP_ImpedanceControllerService_getObjectForceMoments.srv
```
というのを見つけて（どこかで自動生成したのが残ったのだと思いますが）、rtmros_commonごと
`git clean -xfd`
してビルドしなおしたところ、今度は`*ri*`が作れるようになりました。

idl等が変わった時には`--force-cmake`をつけてビルドするだけでなく、gitのリポジトリを綺麗にすると解決することもあることがわかりました。
一応報告しておきます。

snozawa, 2017-06-26:
報告ありがとう。
これはこちらでも際限できるか確認してみます。
もしそうなら、ユーザの手を煩わせず長期的にはCMakeLists.txtで勝手にいいかんじにしてくれるべきなので、調べてみます。

これに加え、ロボットをアップデートしたときに全ユーザに影響がいかないように、
PR2ライクに、wsをわける、ユーザをわけるを各ロボットで使えるように整備しようと思ってます。
（が、こちらは７月くらいになるかも）

mttamtam, 2017-06-26:
srvファイルが残っているのが悪い、と断定してしまったものの、いま問題なく動いていたデスクトップPCのほうを見てみると、
```
tamura@aconcagua:~/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge (master)$ ls srv
Img_CameraCaptureService_start_continuous.srv                            OpenHRP_ImpedanceControllerService_waitImpedanceControllerTransition.srv                    OpenHRP_SequencePlayerService_setJointAnglesSequenceFull.srv
Img_CameraCaptureService_stop_continuous.srv                             OpenHRP_KalmanFilterService_getKalmanFilterParam.srv                                        OpenHRP_SequencePlayerService_setJointAnglesSequenceOfGroup.srv
Img_CameraCaptureService_take_one_frame.srv                              OpenHRP_KalmanFilterService_resetKalmanFilterState.srv                                      OpenHRP_SequencePlayerService_setJointAnglesSequenceWithMask.srv
OpenHRP_AccelerationFilterService_getAccelerationFilterParam.srv         OpenHRP_KalmanFilterService_setKalmanFilterParam.srv                                        OpenHRP_SequencePlayerService_setJointAnglesWithMask.srv
OpenHRP_AccelerationFilterService_resetFilter.srv                        OpenHRP_NullService_echo.srv                                                                OpenHRP_SequencePlayerService_setMaxIKError.srv
OpenHRP_AccelerationFilterService_setAccelerationFilterParam.srv         OpenHRP_OGMap3DService_clear.srv                                                            OpenHRP_SequencePlayerService_setMaxIKIteration.srv
OpenHRP_AutoBalancerService_adjustFootSteps.srv                          OpenHRP_OGMap3DService_getOGMap3D.srv                                                       OpenHRP_SequencePlayerService_setTargetPose.srv
OpenHRP_AutoBalancerService_emergencyStop.srv                            OpenHRP_OGMap3DService_save.srv                                                             OpenHRP_SequencePlayerService_setWrenches.srv
OpenHRP_AutoBalancerService_getAutoBalancerParam.srv                     OpenHRP_ObjectContactTurnaroundDetectorService_checkObjectContactTurnaroundDetection.srv    OpenHRP_SequencePlayerService_setZmp.srv
OpenHRP_AutoBalancerService_getFootstepParam.srv                         OpenHRP_ObjectContactTurnaroundDetectorService_getObjectContactTurnaroundDetectorParam.srv  OpenHRP_SequencePlayerService_waitInterpolation.srv
OpenHRP_AutoBalancerService_getGaitGeneratorParam.srv                    OpenHRP_ObjectContactTurnaroundDetectorService_getObjectForcesMoments.srv                   OpenHRP_SequencePlayerService_waitInterpolationOfGroup.srv
OpenHRP_AutoBalancerService_getGoPosFootstepsSequence.srv                OpenHRP_ObjectContactTurnaroundDetectorService_setObjectContactTurnaroundDetectorParam.srv  OpenHRP_ServoControllerService_addJointGroup.srv
OpenHRP_AutoBalancerService_getRemainingFootstepSequence.srv             OpenHRP_ObjectContactTurnaroundDetectorService_startObjectContactTurnaroundDetection.srv    OpenHRP_ServoControllerService_getDuration.srv
OpenHRP_AutoBalancerService_goPos.srv                                    OpenHRP_ReferenceForceUpdaterService_getReferenceForceUpdaterParam.srv                      OpenHRP_ServoControllerService_getJointAngle.srv
OpenHRP_AutoBalancerService_goStop.srv                                   OpenHRP_ReferenceForceUpdaterService_setReferenceForceUpdaterParam.srv                      OpenHRP_ServoControllerService_getJointAngles.srv
OpenHRP_AutoBalancerService_goVelocity.srv                               OpenHRP_ReferenceForceUpdaterService_startReferenceForceUpdater.srv                         OpenHRP_ServoControllerService_getMaxTorque.srv
OpenHRP_AutoBalancerService_releaseEmergencyStop.srv                     OpenHRP_ReferenceForceUpdaterService_stopReferenceForceUpdater.srv                          OpenHRP_ServoControllerService_getSpeed.srv
OpenHRP_AutoBalancerService_setAutoBalancerParam.srv                     OpenHRP_RemoveForceSensorLinkOffsetService_dumpForceMomentOffsetParams.srv                  OpenHRP_ServoControllerService_getTemperature.srv
OpenHRP_AutoBalancerService_setFootSteps.srv                             OpenHRP_RemoveForceSensorLinkOffsetService_getForceMomentOffsetParam.srv                    OpenHRP_ServoControllerService_getTorque.srv
OpenHRP_AutoBalancerService_setFootStepsWithParam.srv                    OpenHRP_RemoveForceSensorLinkOffsetService_loadForceMomentOffsetParams.srv                  OpenHRP_ServoControllerService_getVoltage.srv
OpenHRP_AutoBalancerService_setGaitGeneratorParam.srv                    OpenHRP_RemoveForceSensorLinkOffsetService_removeForceSensorOffset.srv                      OpenHRP_ServoControllerService_removeJointGroup.srv
OpenHRP_AutoBalancerService_startAutoBalancer.srv                        OpenHRP_RemoveForceSensorLinkOffsetService_setForceMomentOffsetParam.srv                    OpenHRP_ServoControllerService_servoOff.srv
OpenHRP_AutoBalancerService_stopAutoBalancer.srv                         OpenHRP_RobotHardwareService_addJointGroup.srv                                              OpenHRP_ServoControllerService_servoOn.srv
OpenHRP_AutoBalancerService_waitFootSteps.srv                            OpenHRP_RobotHardwareService_calibrateInertiaSensor.srv                                     OpenHRP_ServoControllerService_setJointAngle.srv
[... cut ...]
OpenHRP_ImpedanceControllerService_startObjectTurnaroundDetection.srv    OpenHRP_SequencePlayerService_setJointAngles.srv                                            OpenHRP_WavPlayerService_playWavNoWait.srv
OpenHRP_ImpedanceControllerService_stopImpedanceController.srv           OpenHRP_SequencePlayerService_setJointAnglesOfGroup.srv                                     SetSensorTransformation.srv
OpenHRP_ImpedanceControllerService_stopImpedanceControllerNoWait.srv     OpenHRP_SequencePlayerService_setJointAnglesSequence.srv

```

となっていて古いのが残っていますが問題になっていなさそうです。
となると、git cleanした時に他に消えたファイルのほうが問題になっていたような気がしてきました。

mmurooka, 2017-06-26:
これと全く同じ問題が，hrp2017vのPCのhimでも起きていて，git cleanして解決していました．