start-jsk/rtmros_common #1017: wait-interpolationが遅い

mttamtam, 2017-07-05:
環境に手足を押し付けていくということをするときに、やたら遅いなと思って、調べたところ、例えば
```
(dotimes (i 100)
(send *ri* :angle-vector (send *ri* :state :potentio-vector) 5)
(send *ri* :wait-interpolation))
```
としたときに、理想的には1秒以内に処理が終わって欲しいのですが、wait-interpolationが一回あたり1秒ぐらいかかっていて、数分かかります。
回避方法はありますか？

YoheiKakiuchi, 2017-07-05:
かかっているのはwait-interpolationではないと思います．

https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L347-L348
~~~
(:angle-vector
 (av &optional (tm nil) (ctype controller-type) (start-time 0) &key (scale 1) (min-time 1.0) (end-coords-interpolation nil))
~~~
 となっているので，1秒以下の動作が暗黙で送れないようになっています．
 （ちょっとデフォルトのmin-timeが長すぎるように感じています）

~~~
(send *ri* :angle-vector av 5 :default-controller 0 :min-time 0.0)
~~~
とするとtmどおりに動くと思います．

その場合でも，:angle-vector と :wait-interpolation で 0.2sec ほどかかっているかと思います．
(tm = 1000にしたときに，1.2秒ぐらいかかる)
これも，問題になる場合は本当にどこに時間がかかっているか調べる必要があります．

仕組みは，送るとき
~~~
*ri* -- (ros/action) -> HrpsysSeqStateROSBridge -- (RTM service-call) ->  seq(hrpsys)
~~~

waitが返ってくるとき
~~~
seq(hrpsys) <- (RTM service-call による polling? ) -- HrpsysSeqStateROSBridge -- (ros/action) --> *ri*
~~~

mttamtam, 2017-07-05:
なるほど、min-timeとdefault-controller ?を設定するとだいぶ速くなりました。
そして、ご指摘のように0.2秒ぐらい遅くなってます。

ところで、 :wait-interpolation のかわりに :wait-interpolation-seq を用いても待てるぽいですね。
そして、
```
(dotimes (i 100)
  (send *ri* :angle-vector (send *ri* :state :potentio-vector) 10 :default-controller 0 :min-time 0.0)
  (send *ri* :wait-interpolation-seq)
)
```
とすると、全体で5秒ぐらいで終わるようになりました。
wait-interpolation-seqはload-pattern や angle-vector-sequence-fullのときだけ使うものだと思っていたんですが、普通のwait-interpolationの代わりにも使って良いのでしょうか？

YoheiKakiuchi, 2017-07-06:
:wait-interpolation と :wait-interpolation-seq の違いは、まずは、定義されているクラスが違います。

:wait-interpolation は robot-interfaceクラスに定義されていて、ROS仕様のJointTrajectoryActionにgoalを送って、その結果を待っています。

:wait-interpolation-seq は rtm-ros-robot-interfaceに定義されていて、hrpsysのseqのwaitinterpolationサービスを呼び出しています。

rtm-ros-robot-interfaceを使うロボット(hrpsys-ros-bridge-tutorialsに入っているロボット)は、:angle-vector を送ると最終的にseqに書き込んで動くことになるので、seqのinterpolationを待つことが :wait-interpolation と同じ動きをするように見えます。

現状そうなっていますが、それが必ずしもそれが続いていくとは限らないので、以下の対応を守りましょう。
~~~
:angle-vector -> :wait-interpolation
:angle-vector-sequence-full -> :wait-interpolation-seq
~~~

mttamtam, 2017-07-07:
ありがとうございます。
:angle-vectorのときは:wait-interpolationを使うのが正しい対応関係ということで、:angle-vectorを送って補完を待つときには0.2秒程度の遅延が発生するのを待たないといけないのですね。
できればもっと速くしたいので別の方法を考えてみます。

ところで、:angle-vectorを送って:wait-interpolationをするのを交互にしていると
```
[ WARN] [1499411828.376133208, 82.139000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X257a\
8c18> received wrong goal
[ WARN] [1499411828.377742686, 82.139000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X2579\
0de0> received wrong goal
[ WARN] [1499411828.379215509, 82.139000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X2582\
1b88> received wrong goal
[ WARN] [1499411828.380767219, 82.139000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X257e\
2ff0> received wrong goal
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.045561) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.045865) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.045627) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.046365) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.0458) violate min-angle(0.0)                                                                                
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.045829) violate min-angle(0.0)                                                                              
[ WARN] [1499411829.494499324, 82.985000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #Xcbf4\
cc0> received wrong goal
[ WARN] [1499411829.495853747, 82.985000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #Xcbef\
ec0> received wrong goal
[ WARN] [1499411829.497215595, 82.985000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #Xcce5\
350> received wrong goal
[ WARN] [1499411829.498570586, 82.985000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #Xccfd\
7b0> received wrong goal
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.046537) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.045651) violate min-angle(0.0)                                                                              
;; #<rotational-joint #Xd401fe8 CHEST_JOINT1> :joint-angle(-0.046022) violate min-angle(0.0)                                                                              
[ WARN] [1499411829.824827476, 83.239000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X1115\
b838> received wrong goal
[ WARN] [1499411829.826242307, 83.239000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X1115\
44a0> received wrong goal
[ WARN] [1499411829.828209211, 83.239000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X1167\
62c0> received wrong goal
[ WARN] [1499411829.830494356, 83.251000000]: [fullbody_controller/follow_joint_trajectory_action] feedback-cb #<control_msgs::followjointtrajectoryactionfeedback #X1166\
d038> received wrong goal

```
というのが出てきて、 received wrong goalというのはなぜ出てくるんでしょうか。

YoheiKakiuchi, 2017-07-07:
> :angle-vectorを送って補完を待つときには0.2秒程度の遅延が発生するのを待たないといけないのですね。

これだけど、:wait-interpolation-seq だと 0.2秒の遅延は発生しないのかな？

mttamtam, 2017-07-07:
:wait-interpolation-seq だと遅延しないですね、スムーズに動きます。

YoheiKakiuchi, 2017-07-07:
> :wait-interpolation-seq だと遅延しないですね、スムーズに動きます。

具体的に、wait-interpolationが 0.2秒として、 :wait-interpolation-seq は 何秒なのかな？
それと、waitに時間がかかっていそうという推定はどうやってしているのかな？
トータル時間しか分からない気もする気けど。

mttamtam, 2017-07-07:
トータル時間だけ見てます。
wait-interpolation-seqを使った時に、理想的な時間とほとんど変わらないのに対して、wait-interpolationを使って遅れるというのは、情報がROSの世界をぐるっと回ってきて補完が完了したというのを知るのにそのくらいかかるのだろうというふうに思いますが。

YoheiKakiuchi, 2017-07-08:
https://github.com/jsk-ros-pkg/jsk_roseus/pull/528
これでwait-interpolationとwait-interpolation-seqがほぼ同じになるのではないかと思います。

wait-interpolationがコンマ数秒遅延するとして、不具合があるようだと、プログラムの書き方としてはちょっとちがうような気がしていて、
何をしようとしていて、どういうプログラムになっているのか相談してもらえるといいかと思います。

mttamtam, 2017-07-09:
ありがとうございます。

今は片足立ちをして、遊脚側の足裏についてインピーダンス（正確にはアドミタンス）制御を入れた状態で丁寧に下ろしていって接地させる、ということをしています。
angle-vectorは確かに数十ミリ秒〜数百ミリ秒の処理を念頭に置いていない気もしますが、一方でangle-vector-sequence-fullやload-patternは時系列的にモーションとしての性質を持たせるためにangle-vectorのほかにZMPやらroot-coordsやらoptional-dataを用意しないといけなかったりするので中間ぐらい軽量のangle-vectorを渡せるものがあるといいなと思いました。
本当はRTCを作ってhrpsysのほうで処理を書いたほうがいいのかもしれませんが、以前に @snozawa さんにまずeusで試すことを勧められたのでやってみてました。

snozawa, 2017-07-10:
>  angle-vectorは確かに数十ミリ秒〜数百ミリ秒の処理を念頭に置いていない気もしますが、一方でangle-vector-sequence-fullやload-patternは時系列的にモーションとしての性質を持たせるためにangle-vectorのほかにZMPやらroot-coordsやらoptional-dataを用意しないといけなかったりするので中間ぐらい軽量のangle-vectorを渡せるものがあるといいなと思いました。

これは`:angle-vector-sequcen`といメソッドがあるので、使ってみると良いと思います。
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L405

> 本当はRTCを作ってhrpsysのほうで処理を書いたほうがいいのかもしれませんが、以前に @snozawa さんにまずeusで試すことを勧められたのでやってみてました。

これは何だっけ？
そろそろなれてきたと思うので、必要なものは必要なレイヤで実装して試すといいですね。

> 今は片足立ちをして、遊脚側の足裏についてインピーダンス（正確にはアドミタンス）制御を入れた状態で丁寧に下ろしていって接地させる、ということをしています。

（僕が目的を理解できてるか自信ないですが）地面に足を下ろすのは、:angle-vectorを一回おくるので動くような気がします。

mttamtam, 2017-07-12:
> これは:angle-vector-sequcenといメソッドがあるので、使ってみると良いと思います。
なるほど、angle-vector-sequence-fullのfullとは・・・と思ってました。
今回は、なるべく早く（できればROS Bridgeの動く100Hzぐらい？）動かして力センサを読み取って次の動かし方を決めるということをしたかったので、sequenceはどちらにしても使わないなと言ってから思ったのと、逆に一回だけ:angle-vectorを送るのだとconfiguration-spaceでの補間となり意図しない挙動になるので目的にそぐわないという状況でした。

> そろそろなれてきたと思うので、必要なものは必要なレイヤで実装して試すといいですね。
RTCがどの順番にどこにつながっているかや、一部のは毎周期呼ばれるわけではないらしい（？）などの部分を把握できていないので、またお尋ねさせていただきます。

closed 2017-07-12 by mttamtam