start-jsk/rtmros_common #1018: :off-force-vector を見に行くとエラーが出ることがある

mttamtam, 2017-07-07:
```
Call Stack (max depth: 20):
  0: at (elt (send *ri* :off-force-vector :lleg) 2)
  1: at (abs (elt (send *ri* :off-force-vector :lleg) 2))
  2: at (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2)))
  3: at (if (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (progn (warn "~%Detected.~%") (return nil)))
  4: at (when (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil))
  5: at (while (< i #:dotimes63627) (send *robot* :lleg :move-end-pos (float-vector 0 0 -1) :world) (send *ri* :angle-vector (send *robot* :angle-vector) 50 :default-cont\
roller 0 :min-time 0.0) (send *ri* :state) (send *ri* :wait-interpolation) (when (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return n\
il)) (setq i (1+ i)))
  6: at (let ((i 0) (#:dotimes63627 max-trial)) (declare (integer i #:dotimes63627)) (while (< i #:dotimes63627) (send *robot* :lleg :move-end-pos (float-vector 0 0 -1) :\
world) (send *ri* :angle-vector (send *robot* :angle-vector) 50 :default-controller 0 :min-time 0.0) (send *ri* :state) (send *ri* :wait-interpolation) (when (< 10 (abs (\
elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil)) (setq i (1+ i))) nil)
  7: at (dotimes (i max-trial) (send *robot* :lleg :move-end-pos (float-vector 0 0 -1) :world) (send *ri* :angle-vector (send *robot* :angle-vector) 50 :default-controlle\
r 0 :min-time 0.0) (send *ri* :state) (send *ri* :wait-interpolation) (when (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil)))
  8: at (let ((i 0)) (send *ri* :state) (dotimes (i max-trial) (send *robot* :lleg :move-end-pos (float-vector 0 0 -1) :world) (send *ri* :angle-vector (send *robot* :ang\
le-vector) 50 :default-controller 0 :min-time 0.0) (send *ri* :state) (send *ri* :wait-interpolation) (when (< 10 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn\
 "~%Detected.~%") (return nil))))
  9: at (push-foot)
  10: at (test-friction-estimate)
  11: at (octd)
  12: at #<compiled-code #X54a1430>
/home/tamura/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  no sequence in (elt (send *ri* :off-force-vector :lleg) 2)
```
毎回同じように出るわけではなく、問題なく実行できる時もできない時もありますが、なぜでしょうか。

YoheiKakiuchi, 2017-07-07:
このとき、
~~~
(send *ri* :off-force-vector :lleg)
~~~
は、nil を返しているかな？

それと、これが発生するのは実機 or simulation もしくは、どちらも だろうか？

mttamtam, 2017-07-07:
これはchoreonoidでのシミュレーションでたまに起きています。
次に起きたときに調べてみます。

snozawa, 2017-07-10:
次でたら
```
# 1. hrpsys-baseのrmfo rtcから出力されてるか
rtprint localhost:15005/rmfo.rtc:off_rfsensor -n 1

# 2. rmfoとROSBridgeがつながてるか
rtcat localhost:15005/rmfo.rtc -l

# 3. ROSBridgeからtopicが出力されてるか
rostopic hz /off_rfsensor
```
の３点くらいを調べて見てください。

mttamtam, 2017-07-12:
```
Call Stack (max depth: 20):
  0: at (elt (send *ri* :off-force-vector :lleg) 2)
  1: at (abs (elt (send *ri* :off-force-vector :lleg) 2))
  2: at (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2)))
  3: at (if (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (progn (warn "~%Detected.~%") (return nil)))
  4: at (when (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil))
  5: at (while (< i #:dotimes82957) (send *ri* :state) (send *ri* :angle-vector (send *ri* :potentio-vector) 500 :default-controller 0 :min-time 0.0) (send *ri* :wait-in\
terpolation) (when (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil)) (setq i (1+ i)))
  6: at (let ((i 0) (#:dotimes82957 100)) (declare (integer i #:dotimes82957)) (while (< i #:dotimes82957) (send *ri* :state) (send *ri* :angle-vector (send *ri* :potent\
io-vector) 500 :default-controller 0 :min-time 0.0) (send *ri* :wait-interpolation) (when (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%")\
 (return nil)) (setq i (1+ i))) nil)
  7: at (dotimes (i 100) (send *ri* :state) (send *ri* :angle-vector (send *ri* :potentio-vector) 500 :default-controller 0 :min-time 0.0) (send *ri* :wait-interpolation\
) (when (< 20 (abs (elt (send *ri* :off-force-vector :lleg) 2))) (warn "~%Detected.~%") (return nil)))
  8: at (test-friction-estimate)
  9: at (octd)
  10: at #<compiled-code #X4bc9f08>
/home/tamura/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  no sequence in (elt (send *ri* :off-force-vector :lleg) 2)
6.E1-irteusgl$ send *ri* :off-force-vector :lleg
nil
```

```tamura@aconcagua:~$ rtprint localhost:15005/rmfo.rtc:off_rfsensor -n 1
comp_args: rtprint_reader0?exec_cxt.periodic.type=PeriodicExecutionContext&exec_cxt.periodic.rate=100.0
[178.534000000] [-38.078649022053625, -20.970473501979523, -1300.098134490628, -1.8370291835188224, -8.146206978272339, -1.8577017124230635]
tamura@aconcagua:~$ rtcat localhost:15005/rmfo.rtc -l
rmfo.rtc  Active
  Category       example
  Description    null component
  Instance name  rmfo
  Type name      RemoveForceSensorLinkOffset
  Vendor         AIST
  Version        315.13.0
  Parent         
  Type           Monolithic
 +Extra properties
-Execution Context 0
  State  Stopped
  Kind   Periodic
  Rate   1000.0
  Owner  rmfo
-DataInPort: qCurrent
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:15005/JAXON_RED(Robot)0.rtc:q
-DataInPort: rpy
  dataport.data_type          IDL:RTC/TimedOrientation3D:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:15005/kf.rtc:rpy
-CorbaPort: RemoveForceSensorLinkOffsetService
  port.port_type  CorbaPort
  Interface:
    Instance name  service0
    Type name      RemoveForceSensorLinkOffsetService
    Polarity       Provided
 +Connected to  /localhost:15005/RemoveForceSensorLinkOffsetServiceROSBridge.rtc:RemoveForceSensorLinkOffsetService
  (2 other connections)
-DataInPort: rfsensor
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:15005/JAXON_RED(Robot)0.rtc:rfsensor
[... 49 lines cut ...]
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:15005/JAXON_RED(Robot)0.rtc:lhsensor
-DataOutPort: off_lhsensor
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  flush,new,periodic
  port.port_type              DataOutPort
 +Connected to  /localhost:15005/ic.rtc:lhsensor
 +Connected to  /localhost:15005/rfu.rtc:lhsensor
 +Connected to  /localhost:15005/st.rtc:lhsensor
 +Connected to  /localhost:15005/octd.rtc:lhsensor
 +Connected to  /localhost:15005/log.rtc:rmfo_off_lhsensor
tamura@aconcagua:~$ 
```

```
tamura@aconcagua:~$ rostopic hz /off_rfsensor
subscribed to [/off_rfsensor]
WARNING: may be using simulated time
```
となりました。

YoheiKakiuchi, 2017-07-13:
これは、off_xxsensorがHrpsysSeqStateROSBridgeとつながっていませんね。

シミュレーションですが、以下で試しました。
https://github.com/start-jsk/rtmros_choreonoid/pull/262

手元環境では、100回に1回つながっていませんでした。
タイミング問題なのでcpuの個数/速度によってばらつくかと思います。

原因は rmfo がinitializeされる前にHrpsysSeqStateROSBridgeの接続が行われてしまうことです。
理論的には実機でも起こるのだと思いますが、実機はhrpsysとros_bridgeのlaunchが分かれていて、
実行タイミングにも差があるので発覚しにくいのだろうと思います。

対応としては、以下にrmfoがinitializeされたか待つ部分を入れることかと思います。
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py

closed 2018-11-23 by mttamtam