start-jsk/rtmros_common #1036: Unexpected behaviour after last DEB binaries update on Nextage robot

hectorherrero, 2017-12-18:
Hi, 

after 2017-11-21 Indigo release DEB binaries update we have detected an unexpected behaviour of the robot on trajectory execution. 

We are not sure which packages are involved but it seems that the followings are related:
ros-indigo-hrpsys-ros-bridge: 1.3.2-0 -> 1.4.1-0
ros-indigo-hrpsys-tools: 1.3.2-0 -> 1.4.1-0
ros-indigo-openrtm-tools: 1.3.2-0 -> 1.4.1-0

The detected issue arise when the robot finish executing a trajectory and start executing next one. Some of the joints of the robot jumps to unexpected value and start increasing their value without sense. 

```
[ INFO] [1513583448.837732787, 73.500000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 197[Hz] (exec 1653 Hz, dropped 197)
[el] [73.675000] error limit over LARM_JOINT0(9), qRef=-1.79769e+308, qCurrent=-547.83 , Error=-1.79769e+308 > 0.18 (limit), servo_state = ON, q=-nan
[el] [73.675000] error limit over LARM_JOINT1(10), qRef=-1.79769e+308, qCurrent=-547.84 , Error=-1.79769e+308 > 0.18 (limit), servo_state = ON, q=-nan
[el] [73.675000] error limit over LARM_JOINT2(11), qRef=-1.79769e+308, qCurrent=-549.585 , Error=-1.79769e+308 > 0.18 (limit), servo_state = ON, q=-nan
[el] [73.675000] error limit over LARM_JOINT3(12), qRef=-1.79769e+308, qCurrent=-548.105 , Error=-1.79769e+308 > 0.18 (limit), servo_state = ON, q=-nan
```
[image]

Attached you can find a simple script that can be use for reproducing the error. This can be simply executed launching the following commands:
```
T1: rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
T2: python bug.py
```

[bug.zip](https://github.com/start-jsk/rtmros_common/files/1567067/bug.zip)

If you have problems to reproduce the error do not hesitate to ask for more information.

k-okada, 2017-12-19:
sorry for trouble, we found https://github.com/start-jsk/rtmros_common/pull/1013 produces new trouble. 
plase check if https://github.com/start-jsk/rtmros_common/pull/1037 fixes your problem

hectorherrero, 2017-12-20:
Thank you for the fast response. I have tested #1037 in simulation and seems that the issue has disappeared. We will continue testing in the real robot for assuring that there are not more side effects.

k-okada, 2018-03-15:
This has been released as 1.4.2 of indigo ros/rosdistro#16639