start-jsk/rtmros_common #1048: Joint trajectory is executed immediately even if its time stamp is in future

pazeshun, 2018-07-09:
`(send *ri* :angle-vector (send *robot* :angle-vector) 1000 nil 10)` immediately starts and completes in 1 sec, but it's wrong behaviour.
angle-vector execution should start at 10 secs after commanding because `start-time` is set as 10 secs.

YoheiKakiuchi, 2018-07-09:
HrpsysSeqStateROSBridge ignores start-time.
(It's not only for roseus but also other ros interface. )

pazeshun, 2018-07-09:
> HrpsysSeqStateROSBridge ignores start-time.

Is there any reason for this?
I think the simplest solution is to add sleep in action client callback when `start-time` is enabled, but does this solution harm something?

pazeshun, 2018-07-10:
> the simplest solution is to add sleep in action client callback when start-time is enabled

I found this solution is harmful because `joint_states` stops while sleeping...

emijah, 2018-07-10:
Do you think you can add a countdown feature somewhere?

pazeshun, 2018-07-10:
> Do you think you can add a countdown feature somewhere?

Maybe countdown feature can be added to execution loop.

emijah, 2018-07-10:
That would be the best place to put it.