start-jsk/rtmros_common #1062: catkinではないROSパッケージの取り扱いについて

kyawawa, 2019-04-09:
Please excuse me for writing Japanese.

タイトルの通り，hrpsysやopenhrp3など，非catkin(build_typeがcmake)のROSパッケージの取り扱いについてです．

問題を先にハッキリさせておくと，非catkinのROSパッケージをincludeやlinkするときに，PATHの順番を相当気をつけないとsourceとdebが混ざってしまいます． ref: https://github.com/start-jsk/rtmros_choreonoid/issues/224 https://github.com/start-jsk/rtmros_choreonoid/issues/263
以下，具体例としてhrpsysで話を進めていきます．

原因は分かっていて，aptで入るhrpsysは${catkin_INCLUDE_DIRS}や${catkin_LIBRARY_DIRS}でおそらくほとんどの場合PATHが通ってしまうのに対して (/opt/ros/${ROS_DISTRO}/includeにPATHが通る)，ソースで入れたパッケージに対しては${hrpsys_INCLUDE_DIRS}など固有の名称を使わないとPATHが通らないということに起因します．
このことが引き起こす問題点を並べると以下のとおりです．
- catkin_INCLUDE_DIRSがhrpsys_INCLUDE_DIRSより前に来るとhrpsysがソースで入っていてもaptの方を参照する
- openrtm_aistがaptでのみ入っている時，openrtm_aist_INCLUDE_DIRSがhrpsys_INCLUDE_DIRSよりも前に来るとhrpsysがソースで入っていてもaptの方を参照する (/opt/rosの方にPATHが通るため)
- 問題に気づきにくい (コンパイルエラーならまだ良いが，中身が変わっていて動作が異なったり，リンク先が異なってセグフォで落ちたりする)

これらの問題点に加えて，難しいのは再発防止です．
上述の通りcatkin_INCLUDE_DIRSやopenrtm_aist_INCLUDE_DIRSを前に持ってきてしまうだけで問題が起こるので，今小手先で直してもいつかまた再発すると思われます．

個人的に思いつく解決策は，
- build_typeをcatkinにする (catkin依存になるので現実的では無さそう)
- build_typeがcmakeのパッケージはrosを使ったaptで入れない

くらいなのですが，何か良い方法はありますでしょうか？
今orocos_kdlを見てみたら同じようにbuild_typeがcmakeで，かつrosのaptでも入るようになっているので，諦めるしかないものなのでしょうか．

余談ですが，現在hrpsys_ros_bridgeはaptのhrpsysが無いとビルドが通らないようになっています．
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L104
このあたりに include_directories(${hrpsys_INCLUDE_DIRS})を入れると通ることは確認しています．

YoheiKakiuchi, 2019-07-23:
> 余談ですが，現在hrpsys_ros_bridgeはaptのhrpsysが無いとビルドが通らないようになっています．
> https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/CMakeLists.txt#L104
> このあたりに include_directories(${hrpsys_INCLUDE_DIRS})を入れると通ることは確認しています．

これは、
https://github.com/start-jsk/rtmros_common/pull/1065
ですね。テストを通す必要があります。

Naoki-Hiraoka, 2022-07-10:
この問題は、今はとりあえずrtmbuildで、catkin_INCLUDE_DIRSと<パッケージ名>_INCLUDE_DIRSの順番を調整することで回避する運用になっています。
https://github.com/start-jsk/rtmros_common/issues/1064 https://github.com/start-jsk/rtmros_common/pull/1106

しかし、catkin_INCLUDE_DIRSと<パッケージ名>_INCLUDE_DIRSの順番を調整するだけでは、どのような順番にしても問題が発生してしまうことがわかりました。

- catkin_INCLUDE_DIRSが<パッケージ名>_INCLUDE_DIRSより前に来ると、<パッケージ名>がソースで入っていてもaptの方を参照することがある
  - find_package(catkin REQIRED COMPENTS ...)の中に、ソースで入っているcatkin依存パッケージが一つも無い場合に発生する
- <パッケージ名>_INCLUDE_DIRSがcatkin_INCLUDE_DIRSより前に来ると、find_package(catkin REQIRED COMPENTS ...)の中のcatkin依存パッケージがソースで入っていてもaptの方を参照することがある
  - <パッケージ名>がソースで入っていない場合に発生する

今のrtmbuildの運用だと、後者の問題が発生してしまいます。例えばhrpsys_ros_bridgeで、https://github.com/ros/common_msgs　をソースから入れてメッセージの定義を変更したりすると、エラーが発生します。

catkin_INCLUDE_DIRSと<パッケージ名>_INCLUDE_DIRSの順番を調整するのではなく、https://github.com/start-jsk/rtmros_common/pull/1068/commits/603b6a7cd76d17a0dbd874379f8531970e1cc930 の方法をrtmbuild全体に適用すれば、解決するかもしれません。

Naoki-Hiraoka, 2022-07-11:
この手の変更は経験上、困っている人が僕一人だけのうちはmergeできないということと、rtmbuildには他にもいくつか変えたいところがあるので、当面はhttps://github.com/Naoki-Hiraoka/rtmbuild2 で進めます。使いたい方や、masterへのmergeを希望する方はご連絡ください。