start-jsk/rtmros_common #1063: How to print original data types by rtprint?

ishiguroJSK, 2019-05-24:
For exsample,
`rtmlaunch hrpsys_ros_bridge samplerobot.launch`
and
`rtprint localhost:15005/SampleRobot\(Robot\)0.rtc:servoState`
returns
`rtprint: Type not found: TimedLongSeqSeq`

I guess the reason is the TimedLongSeqSeq is defined in hrpsys/idl/HRPDataTypes.idl (not in OpenRTM something official).
I also want to use the other data types also defined in this idl, so I want to know how to print and debug these data types.

ishiguroJSK, 2019-05-25:
He also encountered same issue.
https://github.com/start-jsk/rtshell_core/issues/10

But I found this is too hard without committing to the OpenRTM official idl "ExtendedDataTypes.idl" or something.

As he said, "hrpsys/idl/HRPDataTypes.idl" is now used as temporary,
https://github.com/fkanehiro/hrpsys-base/commit/a7c8c4aaffe1f46519db83e37830e150262f5a16

but I wonder why there is no way to show user-defined data types, even through ROS can do.

Naoki-Hiraoka, 2023-08-04:
根本的解決ではありませんが、`rtcat -ll` を使うとoriginal型を1回だけprintすることができます。