start-jsk/rtmros_common #1097: robot と self でlimbの順番が違う

YutaKojio, 2020-05-27:
set-ref-force では指定されたlimbは指定された力にし，
それ以外はself(\*ri\*)の referece-force-vector を使うようになっていますが，
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L539-L542
少なくともJAXONでは，robot では (larm rarm lleg rleg) なのに対し，self は (rleg lleg rarm larm) となっているため意図した値がセットできません．

すぐ直せなかったのでとりあえず報告させていただきます．

Naoki-Hiraoka, 2020-05-27:
SampleRobotでも同じ問題が起きています．
ロボットのwrlファイルにForceSensorが書かれている行の順番と，sensorIdが一致していない場合に，発生します．
JAXONの場合は，wrlファイルに上から順にlhsensor, rhsensor, lfsensor, rfsensorの順で書かれていますが，
sensorIdはlhsensorが3, rhsensorが2, lfsensorが1, rfsensorが0となっています．

- hrpsysはforce sensorをwrlファイルのsensorIdの順に並べて扱うため，hrpsysからのトピックが入っている(send \*ri\* :fer-force-vector)はsensorIdの順にlimbが並ぶ．
- eusモデルは，openhrp-export-colladaがsensorIdを見ずに，wrlファイルに出てくる行の順にforcesensorにインデックスをつけ直す(https://github.com/fkanehiro/openhrp3/blob/53d561a288866a9ebf23c2bd07631fe4c5b91594/server/ModelLoader/ColladaWriter.h#L1080 )ため，(send \*robot\* :force-sensors)はwrlファイルに出てくる行の順にlimbが並ぶ

openhrp-export-colladaを直すか，rtm-ros-robot-interfaceがhrpsysからのトピックを受け取る際にlimbの順を適切に並び替える必要があると思われます．

YutaKojio, 2020-05-27:
ありがとう．samplerobotで似たような問題があった気がして探してすぐ見つけられなかったけどどこのissueだったけ？

同じwrlファイルから作ってるので，
openhrp-export-colladaを直す
のほうがよさそうな気がしますね．

Naoki-Hiraoka, 2020-05-27:
privateリポジトリですが，https://github.com/start-jsk/rtmros_hrp2/pull/550#issuecomment-611375233 になります．

YutaKojio, 2020-05-27:
privateロボット特有というわけでもないみたいなので解決するまではここも残しておきます．

Naoki-Hiraoka, 2021-06-10:
原因はopenhrp3ではなくeuscolladaでした. https://github.com/jsk-ros-pkg/jsk_model_tools/pull/236

Naoki-Hiraoka, 2022-04-06:
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/236 で解決しました

closed 2022-04-06 by Naoki-Hiraoka